LibreDWG/libredwg #14: Not working on 2018 CAD files

tonyatseven, 2018-06-01:
I have just compiled the lib, tried to load in a 2018 dwg file, and it outputs the following:

ERROR: This version of LibreDWG is only capable of safely decoding version R13-R2007 (code: AC1012-AC1021) DWG files.
Support for this version is still experimental. We don't decode all objects yet.
It will probably crash and/or give you invalid output.
ERROR: bit_read_BD: unexpected 2-bit code: '11'
ERROR: bit_read_BD: unexpected 2-bit code: '11'
ERROR: bit_read_BD: unexpected 2-bit code: '11'
ERROR: bit_read_BL: unexpected 2-bit code: '11'
Warning: Invalid handle pointer code 14
ERROR: Invalid handle-reference, longer than 4 bytes: 13.12.0
ERROR: Could not read handleref in the header variables section
ERROR: Invalid handle-reference, longer than 4 bytes: 5.6.0
ERROR: Could not read handleref in the header variables section
Warning: dwg_decode_handleref_with_code: Missing obj arg
Segmentation fault: 11

rurban, 2018-06-02:
Yes? This limitation is clear and documented. 
r2010-r2018 support is actively being worked on.

The segfault does not appear on a released tarball, as the library will abort earlier. Only in the git version it will attempt to decode it.

closed 2018-06-02 by rurban

rurban, 2018-06-08:
BTW: It works now with 2018 DWG files