LibreDWG/libredwg #15: The pre-release for windows cannot run

eryar, 2018-06-06:
The pre-release LibreDWG for Windows cannot run, see follow picture:

[image]

eryar, 2018-06-06:
Hi Reini Urban,

I used the depends to check the dwgread.exe, and find it has not dependency with the libredwg-0.dll:
[image]

Other *.exe have the same problem. Do you use the static library for libredwg?

Best Regards,
Shing Liu

rurban, 2018-06-06:
Nope, it should be the dynamic dll. All the tests on appveyor do run correctly.
See `.appveyor.yml` of the build recipe, and `build-aux/appveyor-deploy.bat` for the geneartion of the zip.
I'll check the logs at https://ci.appveyor.com/project/rurban/libredwg/history

eryar, 2018-06-06:
OK.

I will try to compile libredwg on Windows and have a test.

rurban, 2018-06-06:
Yes please, the logs do look ok to me.
I'll try to revive my windows machine in the meantime.

eryar, 2018-06-06:
The 0.4.660 still the same problem and all the *.exe files are the same size:
[image]

commit 906309e4 (2018-06-06): fix windows deploy: not the libtool wrappers

Fixes #15
The windows make install does not put the proper binaries into bin,
so we went and copied it manually. But the wrong ones.
closed 2018-06-06 by rurban