LibreDWG/libredwg #17: dwg2dxf output extra newline in dxf file

eryar, 2018-06-10:
The dwg2dxf output extra newline in dxf file, so cannot open the DXF file generated by dwg2dxf.

rurban, 2018-06-10:
Thanks for the idea, still working on this one.

rurban, 2018-06-10:
I cannot reproduce that. I emit dos '\r\n', because that's what AutoCAD expects.
I also found one empty line for LTYPE 48, which I fixed with 53643aed23b49a06498dae661f8e9ed3f628c8b7

Can you add a short example what you mean?

eryar, 2018-06-11:
[image]

The test.dxf is generated by dwg2dxf, there is a CR, LF for every group code and group value.
These CR,LF should removed.

rurban, 2018-06-11:
Thanks. I think I see CR CR LF there. Right? 
This could be caused by a text write mode to fopen ("w"). Will use "wb" then.

rurban, 2018-06-11:
BTW: If it would be only the the extra \r it would be trivial to fix.
`sed -i 's/\r\r$//' Drawing.dxf`

But the dxf cannot be imported into acad yet, you still have to wait for that, or try --minimal.

rurban, 2018-06-11:
The extra \r on windows should be fixed now.

closed 2018-06-11 by rurban

eryar, 2018-06-12:
Yes, the extra \r on Windows is fixed now.

And I tried to generate dxf by --minimal option, and cannot open the dxf file in AutoCAD.
I find the output dxf file lack of some import group code and its group value, such as group code 10, 20,

rurban, 2018-06-12:
Thanks, still working on DXF. I added now a second DXF option --low #21, to mimic libdxfrw and other DXF libraries and to get closer to a successful import. Something minor is still missing.

eryar, 2018-06-12:
You are welcome.

That's great!