LibreDWG/libredwg #25: Is it possible to give the windows project file?

myluoxz, 2018-06-22:
Missing config.h file?

Currently, the highest support is to reach this version？

myluoxz, 2018-06-22:
windows build method?

rurban, 2018-06-22:
See the following scripts how we build on windows:

* build-aux/smoke.sh: cross-compilation recipes
* build-aux/msys2.bat: msys2 deps
* .appveyor.yml: windows smoker

We only support C99 compilers.

myluoxz, 2018-06-25:
Thank you!
libredwg Does not support 3D format analysis？？？

rurban, 2018-06-25:
What is 3D format analysis? 

We decrypt ACIS data version 1 to the SAT Ascii format, but we have no idea yet about the new version ShapeManager/ACIS 7.0 derived format.
libredwg has no tesselation or UCS transformation utils yet.
Check out gdal for this.

myluoxz, 2018-06-25:
dwg_read_file()  function call,  Console has a lot of information

myluoxz, 2018-06-25:
CAD  3d   model analysis   not support?
libdxfrw  It doesn't seem like it can parse 3D models？

rurban, 2018-06-25:
libdxfrw is not able to decode 3DSOLID,REGION,BODY,SURFACE entities.
libredwg and libdwg can decode 3DSOLID,REGION,BODY entities version 1.

myluoxz, 2018-06-25:
dwg_read_file() function call, Console has a lot of information?

rurban, 2018-06-25:
Summary: LibreDWG deals with DWG and DXF files, not SAT files.
LibreDWG is able to import and export the SAT data in version 1, not in version 2 (yet).

LibreDWG is not able to manipulate any 3D format, nor tesselate it nor do UCS transforms, though that would be certainly required later to be able to display data in paper space or blocks.
But currently all this is a non-goal. We only represent SAT data in ASCII form for 3DSOLID, REGION and BODY. The various SURFACE classes are version 2, so not. 
3D manipulation of that data are goals for other libraries, such as CASCADE, gdal and various 3D CAD packages, such as LibreCAD, FreeCAD or SolveSpace.

rurban, 2018-06-25:
> dwg_read_file() function call, Console has a lot of information?
Is that a question or a statement?

myluoxz, 2018-06-25:
I draw a cube box in cad2014,  libredwg can not be resolved?

rurban, 2018-06-25:
What is cad2014? Do you mean AutoCAD r2014?
LibreDWG can read all elements from r13-r2018, only some objects not as listed in TODO or src/classes.inc

eryar, 2018-06-25:
@rurban Yes, I agree with you that LibreDWG parse the 3d data to ACIS ASCII is good enough. If someone want to display the 3d data in DWG or convert 3d data to other format, they should use toolkit like ACIS SDK or other geometry kernel.

myluoxz, 2018-06-25:
@eryar   LibreDWG parse the 3d data to ACIS   demo path?

myluoxz, 2018-06-26:
@rurban     I just want to parse the 3D model in DWG format

myluoxz, 2018-06-26:
for (i = 0; i < dwg.num_objects; i++)
    {
      Dwg_Entity_LINE *line;
      Dwg_Entity_CIRCLE *circle;
      Dwg_Entity_TEXT *text;
      printf("DWG type : %u\n", dwg.object[i].type);
      switch (dwg.object[i].type)
        {

myluoxz, 2018-06-26:
dwg.object[i].type      !=   DWG_TYPE__3DFACE,  DWG_TYPE_SOLID, DWG_TYPE_VERTEX_3D,  DWG_TYPE_VERTEX_MESH ,   DWG_TYPE_POLYLINE_3D  ....    

libredwg and libdwg can decode 3DSOLID,REGION,BODY entities version 1.

entities version 1.  ?    How to convert？

myluoxz, 2018-06-26:
[image]

rurban, 2018-06-26:
So there are 2 questions, right?
1) Why does the console still outputs a lot of information (stderr), even with dwg.opts = 0.
2) How to convert 3DSOLID data to version 1?

Question 1:
This should not happen for objects (*.o files) using logging.h with dynamic DWG_LOGLEVEL = loglevel. It only happens for static compiled-in loglevel != NONE. There might be some left.

Question 2:
LibreDWG cannot handle version 2 ACIS data yet. I did it by saving the DWG as format r2000, which converts version 2 ACIS data to version 1 I believe. There might be other AutoCAD tricks, you need to ask an AutoCAD specialist. I was one decades ago, but forgot most of it already, and this is very new knowledge.

myluoxz, 2018-06-26:
[image]

[image]

rurban, 2018-06-28:
See the answers to the questions

closed 2018-06-28 by rurban

franky9911, 2021-07-10:
> See the following scripts how we build on windows:
> 
> * build-aux/smoke.sh: cross-compilation recipes
> * build-aux/msys2.bat: msys2 deps
> * .appveyor.yml: windows smoker
> 
> We only support C99 compilers and POSIX environments.
> Pre-compiled Windows binaries are here: https://github.com/LibreDWG/libredwg/releases

I downloaded the latest release 0.12.4-win64, use in VS2017, shows <dwg.h> at line 9023, error C2219: syntax error: type qualifier must be after '*', is there missing a * before restrict? It reads dwg file well if I comment that function.

franky9911, 2021-07-10:
@rurban 
I compiled successfully from source in cygwin64 terminal, but couldn't find libredwg-0.dll, there is cygredwg-0.dll in bin folder and exe files all work(dep. cygwin1.dll), tried to link same snippet with cygwin64's libredwg.dll.a in VS2017, it stuck there at dwg_read_file(), don't know why.