LibreDWG/libredwg #27: Problem reading gradient hatch from files v2004

arturredzko, 2018-06-27:
I have problems decoding hatch with gradient in these files v2004 :

[DWG.zip](https://github.com/LibreDWG/libredwg/files/2142198/DWG.zip)

Can you please take a look if the spec is correct? Strangely the same files converted to version 2000 or 2007 open fine.

rurban, 2018-06-27:
I see, HATCH handle: 0.3.1E9F74 [5] in CascoUrbano.dwg is wrong.
Will check.

arturredzko, 2018-06-28:
Great :) Thanks for fast response.

rurban, 2018-06-28:
I could fix this bug now. It was of course another ODA spec bug with RBG colors.

The colors of gradient HATCH'es are still not fully converted to DXF properly though. There are some special tricks. Will do this later.

arturredzko, 2018-06-28:
Are your commits for rgb colors final, because I applied them fast way but still get corrupted hatch (number of gradient colors is too big and I get corrupted name strings) ?

rurban, 2018-06-28:
Nope, not yet. Still get corrupted 2013 DWGs.

arturredzko, 2018-06-28:
For v2004 I get corrupted hatch only if color has together AcDbColor reference (0x40) and complex color (0x80)

arturredzko, 2018-06-28:
for 2013 I'm not sure about reading has_ds_binary_data. ODA spec says so, but reading this byte corrupts next reading.

rurban, 2018-06-28:
Exactly, this is wrong. Just deleted it by myself some minutes ago. Great analysis

arturredzko, 2018-06-28:
Ok, I figured out where the problem occurs. Here is extracted hatch 0x1E9F74 and screenshot from editor 
[hatch_with_color_ref.zip](https://github.com/LibreDWG/libredwg/files/2145270/hatch_with_color_ref.zip)
Gradient has two colors : reference color DIC6 and white color. To decode the colors correctly you must check the ignored_color_byte of Dwg_HATCH_Color that is really flags like in ENC: (flags & 1) is color.name, (flags & 2) is color.book_name. In this case it's value of 3 so you need to read FIELD_T twice.

Also after tests I think that in ENC decoder if you have (flags & 0x40) then you should ignore (flags & 0x80) because the color value you will read in COMMON_ENTITY_HANDLE_DATA. 

ODA spec is really buggy.

rurban, 2018-06-28:
I'll leave now for holidays for a week, and will incooperate this into the next release. 
I think the 2 strings name and book_name are only for entities, not for objects, but I'll have to check your new example still.

Yes, the color handle is read from COMMON_ENTITY_HANDLE_DATA. There needs to be a check for the flag & 0x40.

rurban, 2018-07-23:
gradient_fills are now stable

closed 2018-07-23 by rurban