LibreDWG/libredwg #28: Decode objects handles map incorrectly

arturredzko, 2018-06-28:
I found another interesting issue with decoding handles and offsets for objects :
[2004.zip](https://github.com/LibreDWG/libredwg/files/2144361/2004.zip)
I may be wrong but it looks like for long handle address ModularChar reading can use more than 4 bytes, up to 8. Although the handles are read, some section pages contains large values that are not fully decoded  and all handle references from header to block controls (layers, etc.) are not matching.

rurban, 2018-06-28:
Yes, I already have this also in some of my other 2004 examples.

rurban, 2018-06-28:
This didn't make it into the next test release 0.4.924, but I'll keep it on my list.

rurban, 2018-07-23:
Suelo urbano022 PGOU Cullera-2004.dwg has now all correct handles.
This was caused by #27, wrong entity color logic.

closed 2018-07-23 by rurban