LibreDWG/libredwg #35: configure failed

BGEray, 2018-08-09:
Hi, I installed all the dependencies the README mentioned, but when I excute "./configure --enable-trace --disable--write --disable-shared", still an error occured, please check it below:

checking for swig... /usr/bin/swig
checking SWIG version... 2.0.12
checking for SWIG library... /usr/share/swig2.0
checking for a Python interpreter with version >= 2.7... python
checking for python... /root/.pyenv/shims/python
checking for python version... 2.7
checking for python platform... linux2
checking for python script directory... ${prefix}/lib/python2.7/site-packages
checking for python extension module directory... ${exec_prefix}/lib/python2.7/site-packages
checking for python2.7... (cached) /root/.pyenv/shims/python
checking for a version of Python >= '2.1.0'... yes
checking for the distutils Python package... yes
checking for Python include path... -I/root/.pyenv/versions/2.7.9/include/python2.7
checking for Python library path...
ac_python_libdir=/root/.pyenv/versions/2.7.9/lib/python2.7/config
ac_python_library=libpython2.7.a
ac_python_soname=libpython2.7.a
use the --enable-frameworked shared library
-L/root/.pyenv/versions/2.7.9/lib/python2.7/config -lpython2.7
checking for Python site-packages path... /root/.pyenv/versions/2.7.9/lib/python2.7/site-packages
checking python extra libraries...  -lpthread -ldl  -lutil
checking python extra linking flags... -Xlinker -export-dynamic
checking consistency of all components of python development environment... no
configure: error: in `/home/odoo/new/libredwg-0.5.0.1176':
configure: error:
  Could not link test program to Python. Maybe the main Python library has been
  installed in some non-standard library path. If so, pass it to configure,
  via the LDFLAGS environment variable.
  Example: ./configure LDFLAGS="-L/usr/non-standard-path/python/lib"
  ============================================================================
   ERROR!
   You probably have to install the development version of the Python package
   for your distribution.  The exact name of this package varies among them.
  ============================================================================

but I already installed python-dev, and I excute "./configure LDFLAGS="-L/root/.pyenv/versions/2.7.9/lib/python2.7" PYTHON="/root/.pyenv/shims/python" --enable-trace", same error still happed. Here's my config.log, anything wrong about it?
[config.log](https://github.com/LibreDWG/libredwg/files/2273751/config.log)

rurban, 2018-08-09:
Yes, the python probe is still a bit unstable. The official one doesn't work, I improved the one coming with PostgreSQL. For the start you can disable it with --disable-python.

BGEray, 2018-08-09:
OK, I get it, thanks for replying.

rurban, 2018-08-09:
This is clearly an error in the python config.
Your extra libraries miss -lm.

Also, /root/.pyenv/versions/2.7.9/ doesn't look like a python-dev package to me.
python's embedding support is horrible btw.
My probe does look ok. Try without the python LDFLAGS to catch the proper python-dev installation.

BGEray, 2018-08-09:
I'm using pyenv to manage my virtual python environment，anyway，I'll try your advice.
best regards

rurban, 2018-08-11:
Wrong python config

closed 2018-08-11 by rurban