LibreDWG/libredwg #39: Segmentation fault in dwg_model_space_ref()

danfe, 2018-08-30:
I've been converting DWG files made with some quite old version of AutoCAD, like 2004 or even 2002, I'm not sure, and discovered that the library (I've tried recently released v0.6) segfaults on some of them inside `dwg_model_space_ref()` in `dwg.c`, apparently while accessing `dwg->header_vars.BLOCK_RECORD_PSPACE` (stack trace is trivial so I'd waive it).  This happens both on i386 and amd64.

Example of [problematic file](ftp://ftp.tu-graz.ac.at/pub/mind21/e5_experiment/fussabdruck_grundstueck/fu%DFabdruck_plus.dwg).

rurban, 2018-11-06:
Thank, found the problems. 0.6.2 should contain the fixes

closed 2018-11-06 by rurban

danfe, 2018-11-06:
I confirm that these changes fix the problem; the program is now able to process my whole pile of old AutoCAD files without a single segfault.

rurban, 2018-11-07:
Great, will upload 0.6.2 now