LibreDWG/libredwg #40: dxf polyline coordinates

soundd1, 2018-11-02:
Hi, during conversion from dwg to dxf, there is wrong polyline coordinates.  such as : 
Thanks for help.

[testdata.zip](https://github.com/LibreDWG/libredwg/files/2541335/testdata.zip)

ORİGİNAL : 
AcDbPolyline
 90
        4
 70
   129
 43
0.0
 10
-10.0
 20
50.0
 10
-10.0
 20
60.0
 10
0.0
 20
60.0
 10
0.0
 20
50.0

exported by libredwg : 

AcDbPolyline
 90
     4
 70
     1
 10
-10.00000000000000
 20
50.00000000000000
 10
-10.00000000000000
 20
60.00000000000000
 10
0.0
 20
**-10.00000000000000** (must be 60)
 10
0.0
 20
50.00000000000000

rurban, 2018-11-02:
Oops, I'll check, thanks.

rurban, 2018-11-05:
The problem is not in the DXF output, it already reads this from the DWG:
```
points[0]: (-10.000000, 50.000000) [RD 10]
points[1]: (-10.000000, 60.000000) [DD 10]
points[2]: (0.000000, -10.000000) [DD 10]
points[3]: (0.000000, 50.000000) [DD 10]
```

closed 2018-11-05 by rurban