LibreDWG/libredwg #43: double free object_ref item

DenisPryt, 2018-11-19:
In some reasons, 2 pointers refer in single object, and free this object twice.
Object is created in `dwg_decode_handleref_with_code` and stored in `dwg->object_ref` array. Then it return from function and assigned to object's `subentity` field. 
In `dwg_free` at first some object is destroyed with his `subentity`, and then all `object_ref` is destroyed with thouse `subentity` one more time.
Can you help me fix this bug? Maybe you are know how this problem can be solved?

DenisPryt, 2018-11-19:
If you want to dive into the problem, I can send you the problem file

DenisPryt, 2018-11-19:
Second free is invoked from `common_entity_handle_data.spec` file:
```
if (FIELD_VALUE(entity_mode) == 0 || IF_IS_ENCODER)
  {
    FIELD_HANDLE(subentity, 4, 0); // doc: owner ref always?
  }
```

rurban, 2018-11-20:
I'm not sure if we can eliminate all double free's, since we have arbitrary refs around which can always point to an already free'd object. But we can ensure to NULL every free'd object, so it will not be freed again.

DenisPryt, 2018-11-20:
[This file](https://github.com/DenisPryt/libredwg/blob/issue_43_dwg_file/test/test-data/double_free_example.dwg)
You can catch creation moment in `decode.c` file on line `2910 dwg->object_ref[dwg->num_object_refs++] = ref;` when `dwg->num_object_refs` equals 191806

DenisPryt, 2018-11-20:
And first `free` is invoked in `free.c` file on line `678 dwg_free_object(&dwg->object[i]);` (loop for not control objects).
objectIndex is 44637, objectType is 642, objectFixedType is `DWG_TYPE_UNKNOWN_ENT`

rurban, 2018-11-20:
Oops, lotta problem with this file. I can repro it. It's a UNKNOWN_ENT problem, which is rare.

commit b5c56313 (2018-11-20): free: fix double-free with subentity handle in UNKNOWN_ENT

such handles are already freed with object_ref[]
Fixes #43, detected by DenisPryt
closed 2018-11-20 by rurban