LibreDWG/libredwg #44: ERROR: Invalid insert_handles vcount 66. Need min. 528 bits for HANDLE, have 51 for BLOCK_HEADER

turtle0x1, 2018-11-20:
Hi compiled and ran Libredwg and am trying to read some of our dwgs, I am unfortunately not a machine were I can anonymize the DWG so I can't upload it, does this error make senses without it ?
Drawing was made using Autocad 2016

rurban, 2018-11-20:
Can you email it to me privately? Looks like a structural error somewhere. This is just a symptom from some previous error.

rurban, 2018-11-20:
reini.urban@gmail.com is good

rurban, 2018-11-23:
I think I've fixed this bug, esp. with older r12 DWG's.
You can check the `smoke/eed` branch

rurban, 2018-11-23:
If it still occurs please re-open

rurban, 2018-11-26:
Thanks for the nice sample. The dwg.spec is wrong for BLOCK_HEADER.num_inserts. Currently it's a BL, but it should be a series null-terminated RC's.

rurban, 2018-11-26:
I've also fixed the LTYPE.dash problem now with f3760b5f691295c02514ca1f4cc448f124e9d205, needs a 0.7 bump.

turtle0x1, 2018-11-26:
hey if you run `dwgread test.dwg -O json -o test.json` you will get a segfault it does output the JSON but its incomplete, also dwglayers doesn't output anything on the dwg, want me to open these as separate issues ?

There is also lots of warnings like 

```
Warning: Object handle not found, 65 in 11073 objects
Warning: Object handle not found, 80 in 11073 objects
Warning: Object handle not found, 65 in 11073 objects
Warning: Object handle not found, 80 in 11073 objects
Warning: Object handle not found, 65 in 11073 objects
Warning: Object handle not found, 80 in 11073 objects
Warning: Object handle not found, 65 in 11073 objects
Warning: Object handle not found, 80 in 11073 objects
Warning: Object handle not found, 65 in 11073 objects
Warning: Object handle not found, 80 in 11073 objects
Warning: Object handle not found, 208 in 11073 objects
Warning: Object handle not found, 208 in 11073 objects
Warning: Object handle not found, 82 in 11073 objects
```

When running `dwgread` on the provided drawing but does output success :smile: ill try and loading the resulting dxf into librecad when its done compiling

rurban, 2018-11-27:
JSON is incomplete still, known limitation so far.
for each segfault please a seperate issue, best with the dwg or backtrace.
the object handles not found are just minor limitations.

commit 64a32a88 (2018-11-23): Fix VECTOR_CHKCOUNT segfault

on unassigned obj->dxfname's (esp. for <=R12). Fixes #44.

also rename the error message: vcount is only relevant for vectors, but it
is also used for strings.
commit f3760b5f (2018-11-26): Fix LTYPE.dash, change RC to unsigned char

avoid such negative values. Fixes #44.
This changed the API, thus we must bump to 0.7
commit 6e4cdcf8 (2018-11-26): Fix LTYPE.dash, change RC to unsigned char

avoid such negative values. Fixes #44.
This changed the API, thus we must bump to 0.7
commit 3c3cb828 (2018-11-26): Fix LTYPE.dash, change RC to unsigned char

avoid such negative values. Fixes #44.
This changed the API, thus we must bump to 0.7