LibreDWG/libredwg #55: ERROR: Invalid LTYPE.dash x 61

turtle0x1, 2018-11-27:
When running over some dwgs from the same project I get the errors

```
Warning: Invalid handleref: wanted code 4, got (1.7.0)
ERROR: Invalid LTYPE.dash x 61

ERROR: Invalid LTYPE.dash x 215

ERROR: num_inserts [RC 0]: 136!

Warning: Invalid handleref: wanted code 4, got (1.7.0)
ERROR: Invalid LTYPE.dash x 61

ERROR: Invalid LTYPE.dash x 215
```

rurban, 2018-11-27:
This should have been fixed with ca05a711d94f746dabd7f96c155dbfb58c3de280
Is it not?

turtle0x1, 2018-11-27:
Nope just recompiled master (although i was on the smoke branch) and it doesn't work with at least one of the .dwg want a copy ?

rurban, 2018-11-27:
yes, please

rurban, 2018-12-02:
Will eventually fix later. For now its too harmless

closed 2018-12-02 by rurban