LibreDWG/libredwg #57: Extract Xref Information

turtle0x1, 2018-11-27:
Is it possible to extract xref information from a dwg ?

rurban, 2018-11-27:
You see all the xref info in the INSERT object. You mean you want to extract all xref paths and proces those then recursively. Easy to do. LibreDWG is a library to do that.

turtle0x1, 2018-11-27:
Yeah that would be great if that could be done ?

turtle0x1, 2018-11-29:
@rurban im having ago at trying to implement this, but im getting abit stuck trying to find the INSERT object, could you point me in the direction I need to look ?

rurban, 2018-11-29:
dwg_get_INSERT returns a list of all INSERT objects. it's in dwg_api

rurban, 2018-11-29:
Its generated via a macro GET_DWG_ENTITY(INSERT)

rurban, 2018-11-30:
I've now fixed dwg_get_INSERT to add a NULL-termination, and added it to the docs.
Normal iteration as in all other tools would be faster and recommended though.