LibreDWG/libredwg #64: dwgfilter: XPath-like query expr

rurban, 2018-12-06:
In principle like osmfilter, but with a sane query expression language.
Either XPath or jsonpath wth the dot. notation seem to be the best fit.

By default pure, only display the result of the filter query.
It also needs to handle destructive changes: --delete and --keep for simple boolean queries
and maybe also changes (renames, transform) by assignment ops (+=) or s///.

turtle0x1, 2019-06-25:
Was this released in 0.8 ?

rurban, 2019-06-25:
Nope, sorry
I just released the latest updates, because I'm busy with something else and it will need a few months until the next update.

turtle0x1, 2020-01-17:
Hi @rurban been a while, I still very much need the functionality but am getting build errors when trying to build from https://github.com/LibreDWG/libredwg/commit/24695ce04c43403a152c8899e222157d1d7061a3 any updates or anything I can do to help ?

rurban, 2020-01-17:
The current plan is to use jq as filter with json out | jq | json in.
What's missing is the json importer on which I will start working soon. I already finished the much harder dxf importer, so json will be much easier.

turtle0x1, 2020-01-17:
Hi ruban, 

I have managed to modify the program to get it "working" but the output isn't particularly clear to me (maybe my modifications have broken things) 

First i have disabled the functions 
```
match_MLINESTYLE
match_LAYER_INDEX
match_ACDBNAVISWORKSMODELDEF
```
Then removed their usage  
```
   ...
   else if (obj->type == //the type)
   ...
```
Then renamed these old functions
```
get_first_owned_object -> get_first_owned_entity
get_next_owned_object -> get_next_owned_object
```
Now when I run the program against (test2.dwg) which i previously sent you

like so `./dwgfilter -i 'BLOCK_HEADER.xref_pname' /home/a/libredwg/dantest/test2.dwg` I get no output

When I run the command `./dwgfilter '19' /home/a/libredwg/dantest/test2.dwg`
I get the output 
```
/home/a/libredwg/dantest/test2.dwg MTEXT 1: 1905
/home/a/libredwg/dantest/test2.dwg MTEXT 1: 1195
/home/a/libredwg/dantest/test2.dwg MTEXT 1: 1930
/home/a/libredwg/dantest/test2.dwg BLOCK_HEADER 1: REDACTED_PATH_TO_XREF
/home/a/libredwg/dantest/test2.dwg MTEXT 1: 1985
/home/a/libredwg/dantest/test2.dwg MTEXT 1: 1915
/home/a/libredwg/dantest/test2.dwg MTEXT 1: 1905
/home/a/libredwg/dantest/test2.dwg MTEXT 1: 2190
/home/a/libredwg/dantest/test2.dwg MTEXT 1: 2.191
/home/a/libredwg/dantest/test2.dwg MTEXT 1: SOME_REDACTED_FILE_NAME
```
Now '19' is a line entity according to `dwg.h` so im confused as to why its finding lots of MTEXT ? 

Thanks
Dan

turtle0x1, 2020-01-17:
Ill take a look as your most recent commit when i get home, your commit style is abit odd so its difficult to really see whats changed

turtle0x1, 2020-01-17:
@rurban  still getting tons of build errors on this new version

turtle0x1, 2020-01-20:
d0b9def - yay no build errors :smile: 

It still doesn't quite work as described: 
> Now when I run the program against (test2.dwg) which i previously sent you
> 
> like so ./dwgfilter -i 'BLOCK_HEADER.xref_pname' /home/a/libredwg/dantest/test2.dwg I get no output
> 
> When I run the command ./dwgfilter '19' /home/a/libredwg/dantest/test2.dwg
> I get the output
> 
> /home/a/libredwg/dantest/test2.dwg MTEXT 1: 1905
> /home/a/libredwg/dantest/test2.dwg MTEXT 1: 1195
> /home/a/libredwg/dantest/test2.dwg MTEXT 1: 1930
> /home/a/libredwg/dantest/test2.dwg BLOCK_HEADER 1: REDACTED_PATH_TO_XREF
> /home/a/libredwg/dantest/test2.dwg MTEXT 1: 1985
> /home/a/libredwg/dantest/test2.dwg MTEXT 1: 1915
> /home/a/libredwg/dantest/test2.dwg MTEXT 1: 1905
> /home/a/libredwg/dantest/test2.dwg MTEXT 1: 2190
> /home/a/libredwg/dantest/test2.dwg MTEXT 1: 2.191
> /home/a/libredwg/dantest/test2.dwg MTEXT 1: SOME_REDACTED_FILE_NAME
> 
> Now '19' is a line entity according to dwg.h so im confused as to why its finding lots of MTEXT and an XREF path ?

turtle0x1, 2020-02-13:
@rurban still semi frequently building this, it still doesn't quite work as expected in-regards to extracting xref names im using the command `dwgfilter 'BLOCK_HEADER.xref_pname' DWG_PATH` but it returns nothing

doing something silly like `dwgfilter '1' DWG_PATH` returns lots of items, some of which are xref paths which start with `BLOCK_HEADER 1:`, if there is anything I can do to help (development (if you can point me in the right direction) or more drawings) I would be happy todo so!

rurban, 2020-02-13:
the current filter branch just has grep. the new filter will work via out json, jq, in json.
Just working on that now. It's getting closer on r2000, just tables and fields are broken still.

turtle0x1, 2020-02-13:
Can you hint at how id get xref names ? Thanks again 😊

turtle0x1, 2020-02-13:
@rurban this produces invalid json that python cant read, trailing values like  `"value": 3.` aren't valid

rurban, 2020-02-13:
> @rurban this produces invalid json that python cant read, trailing values like `"value": 3.` aren't valid

Still working on json...

turtle0x1, 2020-02-14:
@rurban my coding skills in C are 0, For me it comes from the `"type": 69` "DIMSTYLE" - thats the only errors in my json so far - I spent a few hours trying to fix it but I give up!

rurban, 2020-02-15:
DIMSTYLE type was broken indeed. Fixed with c21b6fff3f856f36c4938b6eec6f1d4bdabb8660

closed 2020-02-15 by rurban

turtle0x1, 2020-02-15:
@rurban that didn't fix it :disappointed:

rurban, 2020-02-15:
Then I'd love to get that DWG

turtle0x1, 2020-02-24:
@rurban whats your email ?