LibreDWG/libredwg #73: `out_json.c` produces incomplete entity information

nick-bull, 2018-12-13:
Certain entities, e.g., `LINE`, don't have all of their relevant properties mapped, e.g., `insertion_pt`, `start_pt`, `end_pt` into json: 

    {
      "entity": "LINE",
      "type": 19,
      "size": 60,
      "bitsize": 443,
      "picture_exists": 0,
      "thickness": 0.000000,
      "extrusion.x": 0.000000,
      "extrusion.y": 0.000000,
      "extrusion.z": 1.000000    
    }

Any ideas how to extend the data? I can't work out how to.

nick-bull, 2018-12-13:
Another argument with some malformed outputs (note `edge_obscured_color.name`):

    {
      "object": "VISUALSTYLE",
      "type": 506,
      "size": 169,
      "bitsize": 1282,
      "desc": "OverhangOff",
      "type": 21,
      ...
      "edge_obscured_color": 0,
      "edge_obscured_color".rgb: 0000cc,
      "edge_obscured_color.name": "@0	 	A§èG÷`ÈPÚl¢8",
      "edge_obscured_color.bookname": "(null)",
      "parenthandle": "HANDLE(4.1.40) absolute:40",
      "reactors":      [
        "reactors": "HANDLE(4.1.40) absolute:40"
      ],
      "xdicobjhandle": "HANDLE(3.3.94850) absolute:94850"
    },

rurban, 2018-12-14:
With d8dfa350e00baf619e54a33d2f63be0155374065 (smoke/json) the json output is now fixed

commit 7e8b40c3 (2018-12-14): json: use DXF_OR_PRINT blocks (#73)

Most entities only have DXF, DECODER and ENCODER blocks defined (e.g. LINE)
but no special PRINT block, which would be needed for all other
formatter outputs.
For easier spec go with the DXF_OR_PRINT block then, and
define it for those formatters IS_PRINT settings.
This fixes out_json with only IS_PRINT but no IS_DXF action.

We don't want to duplicate DXF {} blocks for PRINT.
And those formatters also go into both directions, in and out.

Thanks to @nick-bull for asking about this.
Closes #73
closed 2018-12-14 by rurban