LibreDWG/libredwg #90: dwg2dxf.exe output dxf invalid?

Febfire, 2019-01-24:
it seems that convert successful:
[image]
but when i open dxf using AutoCAD,got an error,nothing inside.
[image]
my release version : 0.7.1501

Febfire, 2019-01-24:
hi rurban,i retry it wit -v3,but got the same error.
the dwg file just contains a sample polyline.
[image]
just rename file to dwg.
look forward to your reply!

Febfire, 2019-01-25:
hi  rurban,
any progress? @rurban

rurban, 2019-01-25:
It's r14, With the TeighaFileConverter audit I got these:
```
ODA Warning: Database header: Unknown System Variable $DIMSAV
ODA Warning: VPORT: Record name is empty - Ignored
ODA Warning: AcDbLinetypeTableRecord(14): Record name is empty - Ignored
ODA Warning: AcDbLinetypeTableRecord(15): Record name is empty - Ignored
ODA Warning: AcDbLinetypeTableRecord(16): Record name is empty - Ignored
ODA Warning: AcDbLayerTableRecord(10): Record name is empty - Ignored
ODA Warning: AcDbTextStyleTableRecord(11): Record name is empty - Ignored
ODA Warning: AcDbTextStyleTableRecord(132): Record name is empty - Ignored
ODA Warning: AcDbRegAppTableRecord(12): Record name is empty - Ignored
ODA Warning: AcDbRegAppTableRecord(9E): Record name is empty - Ignored
ODA Warning: AcDbRegAppTableRecord(133): Record name is empty - Ignored
ODA Warning: AcDbRegAppTableRecord(134): Record name is empty - Ignored
ODA Warning: AcDbRegAppTableRecord(135): Record name is empty - Ignored
ODA Warning: AcDbRegAppTableRecord(136): Record name is empty - Ignored
ODA Warning: AcDbRegAppTableRecord(165): Record name is empty - Ignored
ODA Warning: AcDbRegAppTableRecord(217): Record name is empty - Ignored
ODA Warning: AcDbRegAppTableRecord(277): Record name is empty - Ignored
ODA Warning: AcDbDimStyleTableRecord(1B0): Record name is empty - Ignored
ODA Warning: AcDbDimStyleTableRecord(137): Record name is empty - Ignored
ODA Warning: AcDbDimStyleTableRecord(27): Record name is empty - Ignored
ODA Warning: AcDbBlockTableRecord(1F): Record name is empty - $TD_AUDIT_GENERATED_0
ODA Warning: AcDbBlockTableRecord(D2): Record name is empty - $TD_AUDIT_GENERATED_0
ODA Warning: AcDbBlockTableRecord(D2): Duplicate record name "$TD_AUDIT_GENERATED_0" - Ignored
ODA Warning: AcDbBlockTableRecord(D6): Record name is empty - $TD_AUDIT_GENERATED_0
ODA Warning: AcDbBlockTableRecord(D6): Duplicate record name "$TD_AUDIT_GENERATED_0" - Ignored
ODA Warning: Invalid owner Object
AcDbPolyline(276)
OdError
```

rurban, 2019-01-25:
Found and fixed the problem. The DXF subclass markers are needed since r13, not just r2000. Check out the new nightly soon. Currently in smoke/dxf