LibreDWG/libredwg #94: Request DWGclean

maadjordan, 2019-02-06:
I would like to thank you for your effort in liberating dwg format.  I would like to request a tool to clean dwg files taking this measures:
1- Remove the thumbnails.
2- replace long path with related path.
3- zerofy embedded GUID.
4- Purge unused items.
5- optimize repeated shapes like "blockify" tool in bricscad.
6- optimize embedded files (like wmf, word,..etc) for size and add option to extract them.
7- optimize compressed streams for size. 

This is what i have in mind and i hope this is hopeful.

Can this be made for DWF format too. We've managed to reduce its size through "Fileoptimizer" and "Leanify".

Regards

rurban, 2019-02-06:
Can you clarify the long path cleanup?
I guess you mean replace long paths to relative paths (eg given on the command-line)
or as with fonts, if the path is in some font-load-path strip it.

maadjordan, 2019-02-06:
you guessed it.

The main goal to anonymize, clean, size reduce dwg files.