LibreDWG/libredwg #95: non elements in features while convert dwg to GeoJSON

Febfire, 2019-02-14:
I convert dwg to GeoJSON using binary 0.7.1600 dwgread.exe.But the output GeoJSON is empty.
[image]

rurban, 2019-02-14:
The currently covered entity types with GeoJSON are:
INSERT, POLYLINE_2D, POLYLINE_3D, LINE, POINT, 
LWPOLYLINE, GEOPOSITIONMARKER

If it's empty your dwg has none of such entities.

Febfire, 2019-02-18:
POLYLINE_3D and MTEXT inside.
[redwg1.zip](https://github.com/LibreDWG/libredwg/files/2874185/redwg1.zip)

Febfire, 2019-02-18:
hi rurban.
I realized that the MTEXT that is not supported causes the parsing to fail.
I look forward to knowing how long I can support the TEXT and MTEXT?

Febfire, 2019-02-18:
It seems that read POLYLINE_3D failed.Just LINE feature inside in output GeoJSON.

rurban, 2019-02-19:
There's no POLYLINE_3D in your example, just a LWPOLYLINE.
I've added now support for the Text in feature for TEXT, MTEXT, INSERT

Febfire, 2019-02-20:
can i extract the color of entity while convrting GeoJSON?

rurban, 2019-02-20:
Not yet. I need to see an example how this fearure is represented. Color as index or rgb?

Febfire, 2019-02-20:
Rgb or hex in need,just like the GeoJSON from dxf using GDAL.But it doesn't support dwg.

rurban, 2019-02-20:
can you point me to an example, how the rgb is encoded.
Now I added color as number for the index, Google maps accepts color names as string.
eg. 
```
"properties": {
        "SubClasses": "AcDbEntity:AcDbLine",
        "Layer": "Tavolo 3",
        "color": 256,
        "EntityHandle": "8B",
        "Text": null
      },
```

Febfire, 2019-02-21:
I used secondary development techniques of AutoCAD to extract color rgb and convert it to hexadecimal.
eg.
(255,255,0) => (#ffff00)
Maybe color index isn't a good choice.I think that the hex is pretty well.

rurban, 2019-02-21:
There many possibilities how colors can be encoded in GeoJSON. We need to pick one.

1. map color with index: int, name: str, rgb: str, alpha: int
eg
```
"color":  {
        "index": 1,
        "rgb": "000007"
      }
```

2. dynamic types to seperate name, rgb or index.
eg `"color": 0, "color": "blue", color: "00ff00"`

3. seperate keys for each type.
eg `"color_index": 0, "color_name": "blue", color_rgb: "00ff00"`

mostly I have no idea of exiting GeoJSON readers, how they accept a color property.

Febfire, 2019-02-21:
I think the second one is a good choice, as of the GeoJSON I have seen.

rurban, 2019-02-21:
I added now variant, but without the names. 
So far I found examples for names pointing to a DBCOLOR handle with something like "DIC 381". 
I believe rgb hex is better for this (rgb": "6CA573")

rurban, 2020-06-23:
I've changed now the hex format to have a leading '#', like '#000000'

commit 4a589165 (2019-02-21): geojson: encode color as rgb (r2004+) or index

Closes #95
closed 2020-06-23 by rurban