LibreDWG/libredwg #99: serveral bugs in LibreDWG

YourButterfly, 2019-03-12:
# libredwg

## version 

    libredwg 0.7 and 0.7.1645

## description

```txt
libredwg
```

## download link

    https://github.com/LibreDWG/libredwg/releases

---------------------

## dwg_dxf_LEADER@dwg.spec:2034-3___null-pointer-dereference

### description

    An issue was discovered in libredwg 0.7 and 0.7.1645, There is a/an null-pointer-dereference in function dwg_dxf_LEADER at dwg.spec:2034-3

### commandline

    dwg2dxf @@ -o /dev/null

### source

```c
None
```

### bug report

```txt

AddressSanitizer:DEADLYSIGNAL
=================================================================
==32285==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000000 (pc 0x7f4d91d2b51e bp 0x0c22000045e3 sp 0x7ffd87ed4b60 T0)
==32285==The signal is caused by a READ memory access.
==32285==Hint: address points to the zero page.
    #0 0x7f4d91d2b51d in dwg_dxf_LEADER /home/pwd/git-fuzz/libredwg/libredwg-0.7/src/./dwg.spec:2034:3
    #1 0x7f4d91d2b51d in dwg_dxf_object /home/pwd/git-fuzz/libredwg/libredwg-0.7/src/out_dxf.c:934
    #2 0x7f4d91ca1ba7 in dxf_entities_write /home/pwd/git-fuzz/libredwg/libredwg-0.7/src/out_dxf.c:1528:18
    #3 0x7f4d91ca1ba7 in dwg_write_dxf /home/pwd/git-fuzz/libredwg/libredwg-0.7/src/out_dxf.c:1596
    #4 0x513785 in main /home/pwd/git-fuzz/libredwg/libredwg-0.7/programs/dwg2dxf.c:255:56
    #5 0x7f4d905aab96 in __libc_start_main /build/glibc-OTsEL5/glibc-2.27/csu/../csu/libc-start.c:310
    #6 0x41a399 in _start (/home/pwd/git-fuzz/libredwg/libredwg-0.7/installed-asan/bin/dwg2dxf+0x41a399)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV /home/pwd/git-fuzz/libredwg/libredwg-0.7/src/./dwg.spec:2034:3 in dwg_dxf_LEADER
==32285==ABORTING

```

### others

    from fuzz project None
    crash name None-00000007-1552381583.dwg
    Auto-generated by pyspider at 2019-03-12 18:15:41

## bit_read_B@___out-of-bounds-read

### description

    An issue was discovered in libredwg 0.7 and 0.7.1645, There is a/an out-of-bounds-read in function bit_read_B at 

### commandline
[... 572 lines cut ...]
=================================================================
==32364==ERROR: AddressSanitizer: SEGV on unknown address 0x00207fff8003 (pc 0x7f4948e0cf48 bp 0x7fffdb01b150 sp 0x7fffdb01aee0 T0)
==32364==The signal is caused by a READ memory access.
    #0 0x7f4948e0cf47 in dwg_dxf_BLOCK_CONTROL /home/pwd/git-fuzz/libredwg/libredwg-0.7/src/./dwg.spec:2154:1
    #1 0x7f4948e0cf47 in dxf_tables_write /home/pwd/git-fuzz/libredwg/libredwg-0.7/src/out_dxf.c:1421
    #2 0x7f4948dce1d5 in dwg_write_dxf /home/pwd/git-fuzz/libredwg/libredwg-0.7/src/out_dxf.c:1589:9
    #3 0x513785 in main /home/pwd/git-fuzz/libredwg/libredwg-0.7/programs/dwg2dxf.c:255:56
    #4 0x7f49476d7b96 in __libc_start_main /build/glibc-OTsEL5/glibc-2.27/csu/../csu/libc-start.c:310
    #5 0x41a399 in _start (/home/pwd/git-fuzz/libredwg/libredwg-0.7/installed-asan/bin/dwg2dxf+0x41a399)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV /home/pwd/git-fuzz/libredwg/libredwg-0.7/src/./dwg.spec:2154:1 in dwg_dxf_BLOCK_CONTROL
==32364==ABORTING

```

### others

    from fuzz project None
    crash name None-00000005-1552381649.dwg
    Auto-generated by pyspider at 2019-03-12 18:15:47

[dwg2dxf.tar.gz](https://github.com/LibreDWG/libredwg/files/2956421/dwg2dxf.tar.gz)

And a same report send to [savannah.gnu.org/bugs](https://savannah.gnu.org/bugs/index.php?55893)

rurban, 2019-03-12:
Fuzzing, nice!

rurban, 2019-04-20:
# dwg_dxf_LEADER@dwg.spec:2034-3___null-pointer-dereference:

Could repro it with 0.7.1645 and master.
This is at FIELD_3DPOINT_VECTOR in LEADER.
There are illegal size fields (num_points) in LEADER on decode, which are not yet set to 0.
```
num_points: 538181910 [BL 76]
ERROR: Invalid points size 538181910. Need min. 3229091460 bits for 3BD, have 1211 for LEADER
num_points: 96 [BL 76]
ERROR: Invalid points size 96. Need min. 576 bits for 3BD, have -1001 for LEADER
```

rurban, 2019-04-20:
# bit_read_B@___out-of-bounds-read

rurban, 2019-04-22:
# dwg_decode_eed_data@decode.c:2353-32___heap-buffer-overflow

This is a logical realloc mismatch in eed idx vs num_eed. leading to internal malloc corruption.

rurban, 2019-04-23:
# dwg_dxf_BLOCK_CONTROL@dwg.spec:2154-1___out-of-bounds-read

added dwg_block_control error handling.
fuzzing created an invalid DWG without initial BLOCK_CONTROL object. Check this and error.
Fixed with d8af38d3bfbf8e85d1cd4ce1695bf3c0bce5fecc

rurban, 2019-04-23:
# bit_convert_TU@bits.c:1323-3___null-pointer-dereference

Check NULL ptr at bit_convert_TU, and bypass STYLE.font_name dxf logic for NULL name.
Fixed with c6f58bcfe1bf07947a0a7ffed267bada19c82920

rurban, 2019-04-24:
All fuzzing vulns fixes now in master, also tested with asan. Thanks a lot.

commit c948548e (2019-04-20): fix #99 first SEGV

dwg_dxf_LEADER@dwg.spec:2034-3___null-pointer-dereference
This is at FIELD_3DPOINT_VECTOR in LEADER.
There are illegal size fields (num_points) in LEADER on decode, which are not yet set to 0.
Add a _LV macro variant for size lvalue, which can be set to 0.

Also found an entity with entmode 3 but empty ownerhandle.
Write an empty ownerhandle then.
Fixed also a typo in writing the mspace block_record with entmode 2: PSPACE => MSPACE
closed 2019-04-24 by rurban