LibreDWG/libredwg #101: building libredwg with conda fails

looooo, 2019-03-26:
I am trying to build libredwg with conda-build but this fails due to python linking:

```
ac_python_libdir=$PREFIX/lib/python3.7/config-3.7m-x86_64-linux-gnu
ac_python_library=libpython3.7m.a
ac_python_soname=libpython3.7m.a
use the --enable-frameworked shared library
-L$PREFIX/lib/python3.7/config-3.7m-x86_64-linux-gnu -lpython3.7
checking for Python site-packages path... $PREFIX/lib/python3.7/site-packages
checking python extra libraries...  -lpthread -ldl  -lutil -lrt
checking python extra linking flags... -Xlinker -export-dynamic
configure: error: in `/opt/conda/conda-bld/libredwg_1553605983551/work':
checking consistency of all components of python development environment... no
configure: error: 
  Could not link test program to Python. Maybe the main Python library has been
  installed in some non-standard library path. If so, pass it to configure,
  via the LDFLAGS environment variable.
  Example: ./configure LDFLAGS="-L/usr/non-standard-path/python/lib"
  ============================================================================
   ERROR!
   You probably have to install the development version of the Python package
   for your distribution.  The exact name of this package varies among them.
  ============================================================================

```

Setting `LDFLAGS` didn't help either. I wonder what's wrong here. I think conda-build needs static python linking. Maybe this is not supported by libredwg?

rurban, 2019-03-26:
Does $PREFIX/lib/python3.7/config-3.7m-x86_64-linux-gnu exist?
The problem is that there exists no stable autoconf python probe. You need to either disable python or provide the proper LDFLAGS. LDFLAGS="-L$PREFIX/lib/python3.7/config-3.7m-x86_64-linux-gnu -lpython3.7m" might work.

looooo, 2019-03-26:
yes exists:

```
ls /opt/conda/conda-bld/libredwg_1553609472014/$PREFIX/lib/python3.7/config-3.7m-x86_64-linux-gnu/  
```
```
config.c  config.c.in  install-sh  libpython3.7m.a  Makefile  makesetup  __pycache__  python-config.py  python.o  Setup  Setup.local
```

Doesn't yet work with this configuration:
```
./configure LDFLAGS="-L$PREFIX/lib/python3.7/config-3.7m-x86_64-linux-gnu -lpython3.7m"
```

I would like to test the python interface. So building without python is not my goal.

rurban, 2019-03-26:
conda certainly is broken. $PREFIX should be /usr/local not a subdir.

looooo, 2019-03-26:
sry, I guess this was my fault by replacing some parts of the path with $PREFIX in a wrong way. I edited my comment.

tadeu, 2019-03-26:
@looooo, could you post your recipe? Also, if possible, a full output from a build. You can also create a "work in progress" (WIP) pull-request for conda-forge, so that the errors may be resolved incrementally.

looooo, 2019-03-26:
good idea:
https://github.com/conda-forge/staged-recipes/pull/8055

rurban, 2019-03-26:
Cool, I see now a bigger picture, conda being a kind of meta CI service for multiple platforms, right? I see what I can do, but for now I can only watch.

Note that current master is not yet packaging worthy, but tests fine. I am in the middle of a rearchitecturing (dynapi) for more import filters (mainly dxf2dwg). This will need at least two months.

looooo, 2019-03-26:
I am currently using version 0.7. 

> Cool, I see now a bigger picture, conda being a kind of meta CI service for multiple platforms, right?

conda is a package-manager. conda-build are the build tools to create conda-packages. And conda-forge is the organisation which makes it possible to use different ci's to build conda-packages.

looooo, 2019-03-26:
for python2 it seems like python was detected the right way. But for python3 also the ci's report the same error.

I am wondering about this line (from the python3 build):
```
use the --enable-frameworked shared library
```

rurban, 2019-03-26:
I added the --enable-frameworked probe to the m4 autoconf-archive probe, because the old one didn't work then. Now I saw someone updated it again. You can try that instead: 
https://github.com/autoconf-archive/autoconf-archive/blob/master/m4/ax_python_devel.m4

looooo, 2019-03-26:
I tried it, but I guess more files need to be updated:

```
CDPATH="${ZSH_VERSION+.}:" && cd . && /bin/sh $SRC_DIR/build-aux/missing aclocal-1.16 -I m4
/home/conda/feedstock_root/build_artifacts/libredwg_1553633354798/work/build-aux/missing: line 81: aclocal-1.16: command not found
WARNING: 'aclocal-1.16' is missing on your system.
```

looooo, 2019-03-27:
@rurban Can you point me to the diff of the configuration files.

rurban, 2019-03-27:
With this error you need to run `autoreconf --install --symlink -I m4` first.

looooo, 2019-03-27:
Ok, I added the install instructions and use latest sources from the master-branch. 
I am still seeing the issue with the python configuration (I deleted the patch which updats ax_python_devel.m4)

Is it possible to update the configure files directly in the master branch?

rurban, 2019-03-27:
Yes, I'm just smoking it...

looooo, 2019-05-07:
good work  :+1: . linux seems to work now (locally)  Now let's see what the ci's say.
https://anaconda.org/freecad/libredwg

rurban, 2019-05-07:
linux seems ok, but this is easy. I'd like to see mingw64 also, because this would be the most popular platform.

looooo, 2019-05-07:
osx is also good. No idea about mingw64. I don't understand how to use autotools for this toolchain. But for me this isn't a show stopper.

rurban, 2020-08-01:
Hi @looooo 
Can you please verify that master as of 4b99edb0ea26e99ef65c5fe68670e6b1f9382d44
works on conda.
This is the 0.11 release candidate.

looooo, 2020-08-01:
https://github.com/conda-forge/libredwg-feedstock/pull/49

seems to work. python bindings for windows are still disabled. also I saw problems after merging previous pr. I will report back once I know more about it.

looooo, 2020-08-14:
@rurban I see some big differences in package-size:
win: ~8mb (no python bindings)
osx: ~24mb
linux: ~52mb

https://anaconda.org/conda-forge/libredwg/files?version=0.11

any ideas why the difference is that big? I guess I should exclude the static library from the package but still the shared-library is about 60.8mb (uncompressed).

looooo, 2020-08-14:
also basic import / export does not work anymore:

```
Warning: dxf2dwg is still experimental.
Reading DXF file /tmp/tmpcelg85pc/test.dxf
Warning: Unknown HEADER.DIMSTYLE STANDARD dxf:3
Warning: Unknown HEADER.CMLSTYLE STANDARD dxf:2
ERROR: Unexpected DXF end-of-file
Warning: Unknown HEADER.DIMSTYLE STANDARD dxf:3
Warning: Unknown HEADER.CMLSTYLE STANDARD dxf:2
Warning: Object handle not found, 1/1 in 31 objects
Warning: Object handle not found, 12/C in 31 objects
Writing DWG file /home/lo/Schreibtisch/test.dwg
```

rurban, 2020-08-14:
Can I have this DXF?

rurban, 2020-08-14:
The size increase is expected. Many more objects were added.

looooo, 2020-08-15:
[test.dxf.zip](https://github.com/LibreDWG/libredwg/files/5078353/test.dxf.zip)

You need to remove the `.zip` extension. (Github doesn't allow to upload dxf).

This is a simple line created with freecad and exported to dxf. I am using this ibredwg version:
```
libredwg                  0.11             py38hbb43d70_0    conda-forge
```

rurban, 2020-08-15:
This cannot be downloaded, because it is passed through zipinfo by the server. Maybe make a real zip for it.

rurban, 2020-08-15:
The problem with this DXF is the missing final EOL in the dxf.
```
  0
EOF
```

If the dxfwriter adds a final "\n" or "\r\n" it will not display the 
ERROR: Unexpected DXF end-of-file

Other than that, this error can be ignored, because the EOF is already assumed, and the conversion goes fine.

Regarding the two missing TEXT and DIMSTYLE's STANDARD warnings: The texts do get the correct handles 71 attached. The warning can be ignored because at that time those handles are not known, and will be resolved later.

The remaining problems are the missing handles for HEADER.DIMSTYLE STANDARD (79)
and HEADER.CMLSTYLE STANDARD, which are harmless, because they are default. Will fix the later.

rurban, 2021-01-31:
For new errors please a new ticket

closed 2021-01-31 by rurban