LibreDWG/libredwg #102: ERROR: Invalid FIELD.childval x 42064384

adipose, 2019-04-09:
dwg2dxf returns "ERROR: Invalid FIELD.childval x 42064384" on a 2010 DWG file.  Not sure what this means.

adipose, 2019-04-09:
Full output below.

```
Warning: Invalid handleref: wanted code 4, got (3.14.0)
Warning: Invalid handleref: wanted code 4, got (3.10.0)
Warning: Invalid handle pointer code 1
Warning: Invalid handleref: wanted code 2, got (1.6.0)
Warning: Invalid handleref: wanted code 4, got (4.8.0)
Warning: Invalid handleref: wanted code 2, got (2.7.0)
Warning: Invalid handleref: wanted code 4, got (2.13.0)
Warning: Invalid handleref: wanted code 4, got (0.12.0)
Warning: Invalid handleref: wanted code 4, got (1.5.0)
Warning: Invalid handleref: wanted code 5, got (3.9.0)
Warning: Invalid handleref: wanted code 3, got (5.15.0)
Warning: Invalid handleref: wanted code 4, got (0.7.0)
Warning: Invalid handleref: wanted code 5, got (3.10.0)
Warning: Invalid handleref: wanted code 3, got (15.12.0)
Warning: Invalid handleref: wanted code 4, got (0.7.0)
Warning: Invalid handleref: wanted code 4, got (0.12.0)
Warning: Unhandled Class object 507 TABLESTYLE (0xfff) 65/0
Warning: Invalid handleref: wanted code 4, got (1.7.0)
Warning: Invalid handleref: wanted code 4, got (4.12.0)
Warning: Invalid handleref: wanted code 5, got (0.8.0)
Warning: Invalid handleref: wanted code 5, got (0.10.0)
Warning: Unhandled Class object 507 TABLESTYLE (0xfff) 105/0
Warning: Unhandled Class object 512 MATERIAL (0x481) 117/0
Warning: Unhandled Class object 512 MATERIAL (0x481) 118/0
Warning: Unhandled Class object 512 MATERIAL (0x481) 119/0
Warning: Invalid handleref: wanted code 4, got (0.12.0)
Warning: Unstable Class object 513 VISUALSTYLE (0xfff) 121/0
Warning: Unstable Class object 513 VISUALSTYLE (0xfff) 122/0
Warning: Unstable Class object 513 VISUALSTYLE (0xfff) 123/0
Warning: Unstable Class object 513 VISUALSTYLE (0xfff) 124/0
Warning: Unstable Class object 513 VISUALSTYLE (0xfff) 125/0
Warning: Unstable Class object 513 VISUALSTYLE (0xfff) 126/0
Warning: Unstable Class object 513 VISUALSTYLE (0xfff) 127/0
Warning: Unstable Class object 513 VISUALSTYLE (0xfff) 128/0
Warning: Unstable Class object 513 VISUALSTYLE (0xfff) 129/0
Warning: Unstable Class object 513 VISUALSTYLE (0xfff) 130/0
Warning: Unstable Class object 513 VISUALSTYLE (0xfff) 131/0
Warning: Unstable Class object 513 VISUALSTYLE (0xfff) 132/0
Warning: Unstable Class object 513 VISUALSTYLE (0xfff) 133/0
Warning: Unstable Class object 513 VISUALSTYLE (0xfff) 134/0
Warning: Unstable Class object 513 VISUALSTYLE (0xfff) 135/0
Warning: Unstable Class object 513 VISUALSTYLE (0xfff) 136/0
Warning: Invalid handleref: wanted code 4, got (12.12.0)
Warning: Invalid handleref: wanted code 4, got (4.12.0)
Warning: Unhandled Class object 510 PLOTSETTINGS (0x0) 140/0
Warning: Unhandled Class object 510 PLOTSETTINGS (0x0) 141/0
Warning: Unhandled Class object 510 PLOTSETTINGS (0x0) 142/0
Warning: Invalid handle pointer code 9
ERROR: Invalid FIELD.childval x 42064384

ERROR: Invalid FIELD.childval x 1074196486
```

adipose, 2019-04-09:
This was compiled on msys2 x64 on Windows.  The binaries from this site also crashed with segfaults.  Also the binary distribution for windows is missing required DLLs which I had to pull from msys2.  When that crashed I decided to build it myself to see if that helped any, and the above was the result.

rurban, 2019-04-19:
This field is known to be unstable, and therefore is ignored in IN_RELEASE builds. Check your config.h and define IN_RELEASE.

The missing msys2 DLL's are annoying, yes. We should really include them in `bin/`

closed 2019-04-26 by rurban