LibreDWG/libredwg #105: null pointer dereference issue

niugx, 2019-04-29:
There is a null pointer dereference issue in dxf_expect_code function, in_dxf.c file, v0.7.1734

Starting program: /libredwg-0.7.1734/programs/.libs/dxf2dwg -v9 ../test/test-data/crafted.dxf
Reading DXF file ../test/test-data/crafted.dxf
TODO: reading DXF not yet done
dxf{0, SECTION}
ERROR: Invalid DXF group code: 269

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff7afd0d3 in dxf_expect_code (dat=0x7fffffffd3c0, pair=0x0, code=2) at in_dxf.c:1088
1088      while (pair->code != code)
(gdb) bt
#0  0x00007ffff7afd0d3 in dxf_expect_code (dat=0x7fffffffd3c0, pair=0x0, code=2) at in_dxf.c:1088
#1  0x00007ffff7b04829 in dwg_read_dxf (dat=0x7fffffffd3c0, dwg=0x7fffffffd4d0) at in_dxf.c:1297
#2  0x00007ffff767f312 in dxf_read_file (filename=0x7fffffffe7fc "../test/test-data/crafted.dxf", dwg=0x7fffffffd4d0) at dwg.c:298
#3  0x0000000000401530 in main (argc=3, argv=0x7fffffffe598) at dxf2dwg.c:225
(gdb) p pair
$1 = (Dxf_Pair * restrict) 0x0
(gdb) 

Guoxiang Niu, EaglEye Team

rurban, 2019-04-29:
Thanks. 
All the importer functions do not work yet, and will be rewritten to use the dynapi very soon.

rurban, 2019-08-06:
See the `smoke/indxf` branch. The old in_dxf is gone soon.

closed 2019-08-06 by rurban