LibreDWG/libredwg #107: nullpointer in out_geojson.c

UmlauteUeberall, 2019-05-22:
I found a nullpointer at out_geojson.c line 391
hdr might be NULL, even if error is not set.
My dirty hack was:

`if (!error && hdr && hdr->type == DWG_TYPE_BLOCK_HEADER)`

the problem lies somewhere in dwg_ref_get_object

UmlauteUeberall, 2019-05-22:
maybe something like this?
dwg_api.c line 21131:
```
dwg_ref_get_object (const dwg_object_ref *restrict ref, int *restrict error)
{
  if (ref)
    {
      *error = 0;
	  if (!ref->obj)
		*error = 1;
      return ref->obj;
    }
  else
    {
      *error = 1;
      LOG_ERROR ("%s: empty ref", __FUNCTION__)
      return NULL;
    }
}
```

rurban, 2019-05-22:
Yes, something like this. Thanks

UmlauteUeberall, 2019-05-22:
sorry, i forgot the brackets

commit 9a99ad7a (2019-06-26): fix NULL in out_geojson

Thanks to @UmlauteUeberall
Closes #107
closed 2019-06-26 by rurban