LibreDWG/libredwg #108: json - nan in GEODATA field

UmlauteUeberall, 2019-05-22:
trying to parse an normal json file I found nan-values in GEODATAs design_pt, using "NaN" should solve this

rurban, 2019-05-22:
I don't quite understand. NaN in the src code somewhere?  NaN appears very often with wrongly parsed objects.

rurban, 2019-07-03:
Or do you mean we should translate nan to "NaN" in json output?

UmlauteUeberall, 2019-07-03:
Exactly, that would solve some parsers problems, since "nan" is often not recognised.

rurban, 2019-07-08:
I found out that I have no single DWG with GEODATA objects to fix this. I just implemented it from scratch. Can you send me one per email?

UmlauteUeberall, 2019-07-11:
I am trying to get free the file I used, sorry for the inconvenience

rurban, 2019-07-17:
You can delete all other entities, I just need a GEODATA object. I'll also not publish it anywhere.

rurban, 2020-03-20:
Fixed this problem for the unstable GEODATA object.
Also disable printing invalid nan values on release versions (i.e. no git checkout)

closed 2020-03-20 by rurban