LibreDWG/libredwg #109: Found null object reference. Could not output an SVG symbol for this BLOCK_HEADER

yoyomule, 2019-06-18:
Hi,
$ dwg2SVG 1.dwg  > 1.svg
Found null object reference. Could not output an SVG symbol for this BLOCK_HEADER
The sample file attached.
[1.zip](https://github.com/LibreDWG/libredwg/files/3303729/1.zip)
Thanks

yoyomule, 2019-06-25:
Thank you, I saw 0.8 just released, I immediately tested it, it seems that the problem still exists, is this considering the scope of repair in the 0.8 version?

[libredwg.zip](https://github.com/LibreDWG/libredwg/files/3327620/libredwg.zip)

rurban, 2019-06-26:
There are two outstanding fixes from the smoke/in-dyn branch, 
dwg_ref_get_object: error on !ref->obj and fix NULL in out_geojson #107 which didn't make it into 0.8.
I was just releasing the tested stuff, which was already used by others. 
0.8 mostly doesn't contain the dynapi being used that much.

but those two and more NULL violations definitely will make it into a 0.8.1, when I have some time. right now I'm too busy with other releases.

rurban, 2019-07-06:
back to it

rurban, 2019-10-02:
Fixed with 6bdfbc1ca4ebf650af8c5e8bb9fa76d45d1cbc56
Some blocks are just empty, ignore them then.

closed 2019-10-02 by rurban