LibreDWG/libredwg #111: Invalid reading of enities with AcDbColor handles

normwill-beast, 2019-07-02:
Hi,

It looks like entities are not read correctly when they include a handle to a color. From the DWG technical specification it looks like the handle does not follow the color flags and index. Below is an extract from the spec:

0x4000: has AcDbColor reference (0x8000 is also set in this case).
The handle to the color is written in the handle stream.

It says that the handle is written in the handle stream, not that it follows...

See attached file with a LWPOLYLINE with color handle

[polyline test.zip](https://github.com/LibreDWG/libredwg/files/3349904/polyline.test.zip)

normwill-beast, 2019-07-03:
Further observation: The reading of the color handle later is also wrong (in common_entity_handle_data.spec). It checks 
   (FIELD_VALUE(color.flag) & 0x40))
It should read 
   (FIELD_VALUE(color.flag) & 0x4000) 
because the value has not be shifted right (like in 'common_entity_data.spec')

rurban, 2019-07-06:
Yes, true. color.handle is wrong, needs to be read from hdl_dat. Just fixed that.
But the flag is seperated from the index, right shifted. so common_entity_handle_data.spec is correct.
The main error is that bit_read_ENC and FIELD_ENC is still not used.
Working on that now

commit 109988d7 (2019-07-06): defer color.handle to handle_stream

for >= R_2007 we dont know the handle stream yet, when reading the
common_entity_data. Defer to the start of the common_entity_handle_data
reading the common color.handle then.
Fixes #111
commit 84bdc444 (2019-07-06): defer color.handle to handle_stream

for >= R_2007 we dont know the handle stream yet, when reading the
common_entity_data. Defer to the start of the common_entity_handle_data
reading the common color.handle then.
Fixes #111, Thanks to normwill-beast.

And while we are there sanitize the common entity color.handle:
No extra color_handle needed. Make the handle a real H (ie. ref* to
the DBCOLOR object).
Remove the obsolete dwg_ent_get_color_handle API.

Extend bit_{read,write}_ENC to include
all 3 streams, but it is still not used yet.
closed 2019-07-06 by rurban