LibreDWG/libredwg #118: FIELDLIST handles (and maybe many more objects) not decoded correctly

normwill-beast, 2019-07-05:
Hi,

The FIELDLIST object does not correctly decode it's field handles because there is an owerhandle, reactors and xdic handle read missing.

The following:
  START_HANDLE_STREAM;
  HANDLE_VECTOR (field_handles, num_fields, 0, 330); // 2 or 4, or 3.0.0

needs to change to (need to add ownerhandle to dwg_object_FIELDLIST 
structure):
  START_HANDLE_STREAM;
  FIELD_HANDLE (ownerhandle, 4, 0);
  REACTORS(4);
  XDICOBJHANDLE(3);
  HANDLE_VECTOR (field_handles, num_fields, 0, 330); // 2 or 4, or 3.0.0

Looking at the dwg.spec it seems to me that EVERY object should have:

  START_HANDLE_STREAM;
  FIELD_HANDLE (ownerhandle, 4, 0);
  REACTORS(4);
  XDICOBJHANDLE(3);
  // Further object handles start here

before reading any other objects. Every object currently reads <num_reactors> and <xdic_missing_flag> so that says to me that these handles may exist at the start of the handles stream. I have tested this with several objects:

FIELD            - Already in issue #117 
FIELDLIST
MLEADERSTYLE - This needs the change I have suggested above

A quick search through dwg.spec for "START_HANDLE_STREAM" showed many inconsistencies. 
For example, BLOCK_CONTROL (indeed all '_CONTROL' objects) had a missing REACTORS(4). Several other objects (such as TABLECONTENT) would start reading private object handles immediately after the START_HANDLE_STREAM.

I propose that a new macro is created called "START_OBJECT_HANDLE_STREAM" which simply expands to 
  START_HANDLE_STREAM;
  FIELD_HANDLE (ownerhandle, 4, 0);
  REACTORS(4);
  XDICOBJHANDLE(3);

and that all objects call this instead of just START_HANDLE_STREAM; 

The only thing I am not sure of - does EVERY object have an ownerhandle.

rurban, 2019-07-05:
START_OBJECT_HANDLE_STREAM sounds good, yes.
I'll start with your proposals next week. Maybe we can generalize it that easily

rurban, 2019-07-07:
Fixed these handles, and generalized START_OBJECT_HANDLE_STREAM as described.
All objects have now a common ownerhandle in obj->tio.object, analog to reactors and xdicobjhandle.

closed 2019-07-07 by rurban

rurban, 2019-07-07:
The following should be done instead:

* setup the handle and string stream before (in dwg_decode_object).
  this could be limited/sliced in range to the specific object only from the main dat stream, to avoid overflows into a next object, and simplify error handling, which is currently object specific. See #85.
* read the handle stream at the beginning, before. (in dwg_decode_object).
  The extra fields as specified can be mixed into the spec, as e.g. in MLINESTYLE, and do not need to be at the very end.
* then read all the object specific fields as specified in the spec, which does not need the boilerplate
START_OBJECT_HANDLE_STREAM at the end anymore.