LibreDWG/libredwg #120: Try to unbreak R2004 decoder on big-endian machines

danfe, 2019-07-13:
Consider attached [patch](https://github.com/LibreDWG/libredwg/files/3388944/decode-endian-fixes.txt).  It might not be perfect and I've only tested it under FreeBSD (`#include <sys/endian.h>`), so some adjustments are most certainly in order for GNU/Linux and other systems, but it seems to fix R2004 decoding and allow to pass those tests.  Patch for R2007 decoder would follow.

rurban, 2019-07-15:
Good idea, but a bit of a hack.
We need to probe for proper endian headers: endian.h sys/endian.h byteorder.h byteswap.h sys/byteorder.h
set the appropriate _POSIX_SOURCE definition for le32toh,le64toh,htole32 to be included ...

danfe, 2019-07-15:
Thank you Reini!  Of course my patch was more like PoC and far from committable state, but I hoped you would straighten out all that portability/testing machinery better than me, and you did. :-)

rurban, 2020-09-04:
I've got now a good CI testing method for a big-endian platform. on travis with s390x. Not really useful imho, as BE is dying, but anyway. See e.g. https://travis-ci.org/github/rurban/safeclib/jobs/724210437

rurban, 2023-04-13:
This is now in since 24d4a79a3a5488e9a3ceae5ce657fe9c3f89d547
Tue Feb 28 2023

closed 2023-04-13 by rurban