LibreDWG/libredwg #126: Several bugs found by fuzzing

strongcourage, 2019-07-27:
Hi,
After fuzzing libredwg, I found the following bugs on the latest commit 959984b on master.
Command: dwg2SVG $PoC

### 1. Invalid free in decode.c:2289
PoC: https://github.com/strongcourage/PoCs/blob/master/libredwg_959984b/PoC_af_decode.c:2289
ASAN says:
~~~
==20736==ERROR: AddressSanitizer: attempting free on address which was not malloc()-ed: 0x7f3e63c329f5 in thread T0
    #0 0x7f3e62c2932a in __interceptor_free (/usr/lib/x86_64-linux-gnu/libasan.so.2+0x9832a)
    #1 0x4973ad in read_2004_section_handles ../../src/decode.c:2289
    #2 0x4a1f6d in decode_R2004 ../../src/decode.c:2425
    #3 0x414259 in dwg_decode ../../src/decode.c:221
    #4 0x4046b8 in dwg_read_file ../../src/dwg.c:193
    #5 0x403767 in main ../../programs/dwg2SVG.c:376
    #6 0x7f3e624de82f in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x2082f)
    #7 0x401948 in _start (/home/dungnguyen/PoCs/libredwg_959984b/dwg2SVG-asan+0x401948)
~~~
Thanks,
Manh Dung

strongcourage, 2019-07-27:
#### 2. Invalid free in dwg.c:197
PoC: https://github.com/strongcourage/PoCs/blob/master/libredwg_959984b/PoC_af_dwg.c:197
ASAN says:
~~~
==17272==ERROR: AddressSanitizer: attempting free on address which was not malloc()-ed: 0x7f2eaf776035 in thread T0
    #0 0x7f2eae72432a in __interceptor_free (/usr/lib/x86_64-linux-gnu/libasan.so.2+0x9832a)
    #1 0x4047b0 in dwg_read_file ../../src/dwg.c:197
    #2 0x403767 in main ../../programs/dwg2SVG.c:376
    #3 0x7f2eadfd982f in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x2082f)
    #4 0x401948 in _start (/home/dungnguyen/PoCs/libredwg_959984b/dwg2SVG-asan+0x401948)
~~~

strongcourage, 2019-07-27:
#### 3. Heap buffer overflow in bits.c:1818
PoC: https://github.com/strongcourage/PoCs/blob/master/libredwg_959984b/PoC_hbo_bits.c:1818
ASAN says:
~~~
==15391==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x7f0f6b847455 at pc 0x0000004132e1 bp 0x7ffe4fb14ab0 sp 0x7ffe4fb14aa0
READ of size 1 at 0x7f0f6b847455 thread T0
    #0 0x4132e0 in bit_calc_CRC ../../src/bits.c:1818
    #1 0x410820 in bit_check_CRC ../../src/bits.c:1213
    #2 0x83b985 in dwg_decode_add_object ../../src/decode.c:4430
    #3 0x489a76 in decode_R13_R2000 ../../src/decode.c:1092
    #4 0x4141ec in dwg_decode ../../src/decode.c:220
    #5 0x4046b8 in dwg_read_file ../../src/dwg.c:193
    #6 0x403767 in main ../../programs/dwg2SVG.c:376
    #7 0x7f0f6a07a82f in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x2082f)
    #8 0x401948 in _start (/home/dungnguyen/PoCs/libredwg_959984b/dwg2SVG-asan+0x401948)
~~~

strongcourage, 2019-07-27:
#### 4. HBO in decode.c:1578
PoC: https://github.com/strongcourage/PoCs/blob/master/libredwg_959984b/PoC_hbo_decode.c:1578
ASAN says:
~~~
==24710==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x62c000007600 at pc 0x00000048ffe9 bp 0x7ffd1c2630c0 sp 0x7ffd1c2630b0
WRITE of size 1 at 0x62c000007600 thread T0
    #0 0x48ffe8 in decompress_R2004_section ../../src/decode.c:1578
    #1 0x493a4c in read_2004_compressed_section ../../src/decode.c:1966
    #2 0x493e3a in read_2004_section_classes ../../src/decode.c:1999
    #3 0x4a1f2f in decode_R2004 ../../src/decode.c:2423
    #4 0x414342 in dwg_decode ../../src/decode.c:226
    #5 0x4046b8 in dwg_read_file ../../src/dwg.c:193
    #6 0x403767 in main ../../programs/dwg2SVG.c:376
    #7 0x7fd8a7f7c82f in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x2082f)
    #8 0x401948 in _start (/home/dungnguyen/PoCs/libredwg_959984b/dwg2SVG-asan+0x401948)
~~~

strongcourage, 2019-07-27:
#### 5. HBO in decode_r2007.c:222
PoC: https://github.com/strongcourage/PoCs/blob/master/libredwg_959984b/PoC_hbo_decode_r2007.c:222
ASAN says: 
~~~
==436==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x62300000de45 at pc 0x00000083d9f7 bp 0x7ffc10e72490 sp 0x7ffc10e72480
READ of size 1 at 0x62300000de45 thread T0
    #0 0x83d9f6 in copy_bytes ../../src/decode_r2007.c:222
    #1 0x843a2b in decompress_r2007 ../../src/decode_r2007.c:539
    #2 0x844255 in read_system_page ../../src/decode_r2007.c:642
    #3 0x845161 in read_sections_map ../../src/decode_r2007.c:810
    #4 0x84e3bd in read_r2007_meta_data ../../src/decode_r2007.c:1673
    #5 0x4a5e7d in decode_R2007 ../../src/decode.c:2463
    #6 0x4142c6 in dwg_decode ../../src/decode.c:222
    #7 0x4046b8 in dwg_read_file ../../src/dwg.c:193
    #8 0x403767 in main ../../programs/dwg2SVG.c:376
    #9 0x7f0484a6e82f in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x2082f)
    #10 0x401948 in _start (/home/dungnguyen/PoCs/libredwg_959984b/dwg2SVG-asan+0x401948)
~~~

strongcourage, 2019-07-27:
#### 6. SBO in dwg.c:1175
PoC: https://github.com/strongcourage/PoCs/blob/master/libredwg_959984b/PoC_sbo_dwg.c:1175
ASAN says:
~~~
==17188==ERROR: AddressSanitizer: stack-buffer-overflow on address 0x7ffcf981e948 at pc 0x00000040aa27 bp 0x7ffcf981e8e0 sp 0x7ffcf981e8d0
WRITE of size 1 at 0x7ffcf981e948 thread T0
    #0 0x40aa26 in dwg_section_type ../../src/dwg.c:1175
    #1 0x845b68 in read_sections_map ../../src/decode_r2007.c:882
    #2 0x84e3bd in read_r2007_meta_data ../../src/decode_r2007.c:1673
    #3 0x4a5e7d in decode_R2007 ../../src/decode.c:2463
    #4 0x4142c6 in dwg_decode ../../src/decode.c:222
    #5 0x4046b8 in dwg_read_file ../../src/dwg.c:193
    #6 0x403767 in main ../../programs/dwg2SVG.c:376
    #7 0x7fa18389382f in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x2082f)
    #8 0x401948 in _start (/home/dungnguyen/PoCs/libredwg_959984b/dwg2SVG-asan+0x401948)
~~~

strongcourage, 2019-07-27:
#### 7. SEGV in bits.c:121
PoC: https://github.com/strongcourage/PoCs/blob/master/libredwg_959984b/PoC_segv_bits.c:121
ASAN says:
~~~
==8815==ERROR: AddressSanitizer: SEGV on unknown address 0x7fa363c1c218 (pc 0x00000040bba4 bp 0x7fff9c760910 sp 0x7fff9c7608f0 T0)
    #0 0x40bba3 in bit_read_B ../../src/bits.c:121
    #1 0x849244 in obj_string_stream ../../src/decode_r2007.c:1224
    #2 0x4aa0a5 in dwg_decode_entity ../../src/decode.c:2929
    #3 0x5a5dc6 in dwg_decode_DIMENSION_DIAMETER_private ../../src/dwg.spec:1231
    #4 0x5a5a37 in dwg_decode_DIMENSION_DIAMETER ../../src/dwg.spec:1231
    #5 0x8387f5 in dwg_decode_add_object ../../src/decode.c:4077
    #6 0x4970de in read_2004_section_handles ../../src/decode.c:2250
    #7 0x4a1f6d in decode_R2004 ../../src/decode.c:2425
    #8 0x414342 in dwg_decode ../../src/decode.c:226
    #9 0x4046b8 in dwg_read_file ../../src/dwg.c:193
    #10 0x403767 in main ../../programs/dwg2SVG.c:376
    #11 0x7fa3590e982f in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x2082f)
    #12 0x401948 in _start (/home/dungnguyen/PoCs/libredwg_959984b/dwg2SVG-asan+0x401948)
~~~

strongcourage, 2019-07-27:
#### 8. SEGV in bits.c:294
PoC: https://github.com/strongcourage/PoCs/blob/master/libredwg_959984b/PoC_segv_bits.c:294
ASAN says:
~~~
==13588==ERROR: AddressSanitizer: SEGV on unknown address 0x7fdf9752e24e (pc 0x00000040cb68 bp 0x7ffcd84aa390 sp 0x7ffcd84aa370 T0)
    #0 0x40cb67 in bit_read_RC ../../src/bits.c:294
    #1 0x41001e in bit_read_H ../../src/bits.c:1120
    #2 0x4b334c in dwg_decode_handleref_with_code ../../src/decode.c:3217
    #3 0x63637b in dwg_decode_BLOCK_CONTROL_private ../../src/dwg.spec:2309
    #4 0x635b00 in dwg_decode_BLOCK_CONTROL ../../src/dwg.spec:2300
    #5 0x838b99 in dwg_decode_add_object ../../src/decode.c:4144
    #6 0x4970de in read_2004_section_handles ../../src/decode.c:2250
    #7 0x4a1f6d in decode_R2004 ../../src/decode.c:2425
    #8 0x414342 in dwg_decode ../../src/decode.c:226
    #9 0x4046b8 in dwg_read_file ../../src/dwg.c:193
    #10 0x403767 in main ../../programs/dwg2SVG.c:376
    #11 0x7fdf9290482f in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x2082f)
    #12 0x401948 in _start (/home/dungnguyen/PoCs/libredwg_959984b/dwg2SVG-asan+0x401948)
~~~

rurban, 2019-07-28:
Nice, thanks

strongcourage, 2019-07-28:
You can fuzz it using AFL. The setup is very simple as follows:
~~~
git clone https://git.savannah.gnu.org/git/libredwg.git; cd libredwg
autoreconf -i
mkdir obj-afl; cd obj-afl
CC=/path/to/afl/afl-gcc ../configure --disable-shared
cd programs; mkdir in; 
cp ../../test/test-data/*.dwg in # should remove some big files > 1 Mb
# fuzz dwg2dxf
/path/to/afl/afl-fuzz -m none -i in -o out_dwg2dxf ./dwg2dxf @@ -o /dev/null
# fuzz dwg2SVG
/path/to/afl/afl-fuzz -m none -i in -o out_dwg2SVG ./dwg2SVG @@ /dev/null
~~~
The reported bugs were easily to be detected after 2 hours, but the triaging step using ASAN was a bit tedious. My suggestion is that you can perform fuzzing to detect some easy-to-catch bugs in your modified code to make `libredwg` more robust. I am also happy to do so. Thanks.

strongcourage, 2019-07-28:
Thanks @rurban for the patches. I rebuilt the program `libredwg` (commit b0b41f5 on branch `smoke/gh126-fuzzing`) with ASAN using your command as mentioned in b0b41f5. 
~~~
CFLAGS="-O2 -g -fsanitize=address -fno-omit-frame-pointer" ./configure --disable-static --disable-bindings --disable-dxf --disable-write
~~~

Then, I run it with crashing inputs generated by fuzzing on the buggy commit  959984b on master. Some bugs have been fixed. However, if you don't mind, I report some bugs that have still existed after your fixes as follows.

Command: ./examples/dwg2svg2 $PoC

#### 9. HBO in bits.c:1826
It seems due to the incomplete fix of #126-3.
PoC: https://github.com/strongcourage/PoCs/blob/master/libredwg_b0b41f5/PoC_hbo_bits.c:1826
ASAN says:
~~~
==28015==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x62a0000057cd at pc 0x7f3c50b90145 bp 0x7ffd882ff990 sp 0x7ffd882ff980
READ of size 1 at 0x62a0000057cd thread T0
    #0 0x7f3c50b90144 in bit_calc_CRC ../../src/bits.c:1826
    #1 0x7f3c50c08c4d in decode_R13_R2000 ../../src/decode.c:932
    #2 0x7f3c50b910d8 in dwg_decode ../../src/decode.c:220
    #3 0x7f3c50b81d78 in dwg_read_file ../../src/dwg.c:193
    #4 0x4019b5 in test_SVG ../../examples/dwg2svg2.c:119
    #5 0x404809 in main ../../examples/dwg2svg2.c:505
    #6 0x7f3c503d982f in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x2082f)
    #7 0x4017e8 in _start (/home/dungnguyen/PoCs/libredwg_1406432/.libs/lt-dwg2svg2+0x4017e8)
~~~

strongcourage, 2019-07-28:
#### 10. HBO in bits.c:294 (#126-8)
PoC: https://github.com/strongcourage/PoCs/blob/master/libredwg_b0b41f5/PoC_hbo_bits.c:294
ASAN says:
~~~
READ of size 1 at 0x62c00000f600 thread T0
    #0 0x7f8f79459367 in bit_read_RC ../../src/bits.c:294
    #1 0x7f8f7945a75f in bit_read_H ../../src/bits.c:1120
    #2 0x7f8f7946b2a3 in dwg_decode_handleref_with_code ../../src/decode.c:3237
    #3 0x7f8f7954736b in dwg_decode_BLOCK_CONTROL_private ../../src/dwg.spec:2310
    #4 0x7f8f7965072c in dwg_decode_BLOCK_CONTROL ../../src/dwg.spec:2300
    #5 0x7f8f796c373d in dwg_decode_add_object ../../src/decode.c:4182
    #6 0x7f8f796c9823 in read_2004_section_handles ../../src/decode.c:2270
    #7 0x7f8f796c9823 in decode_R2004 ../../src/decode.c:2445
    #8 0x7f8f796d195c in dwg_decode ../../src/decode.c:226
    #9 0x7f8f7945387a in dwg_read_file ../../src/dwg.c:193
    #10 0x4019b5 in test_SVG ../../examples/dwg2svg2.c:119
    #11 0x404809 in main ../../examples/dwg2svg2.c:505
    #12 0x7f8f78c1582f in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x2082f)
    #13 0x4017e8 in _start (/home/dungnguyen/PoCs/libredwg_b0b41f5/.libs/lt-dwg2svg2+0x4017e8)
~~~

strongcourage, 2019-07-28:
#### 11. HBO in decode.c:1589 (#126-4)
PoC: https://github.com/strongcourage/PoCs/blob/master/libredwg_b0b41f5/PoC_hbo_decode.c:1579
ASAN says:
~~~
==1662==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x62c00000f600 at pc 0x7fb6378b710d bp 0x7ffebbc956b0 sp 0x7ffebbc956a0
WRITE of size 1 at 0x62c00000f600 thread T0
    #0 0x7fb6378b710c in decompress_R2004_section ../../src/decode.c:1589
    #1 0x7fb6378b817c in read_2004_compressed_section ../../src/decode.c:1986
    #2 0x7fb637b1c6a2 in read_2004_section_handles ../../src/decode.c:2207
    #3 0x7fb637b1c6a2 in decode_R2004 ../../src/decode.c:2445
    #4 0x7fb637b2495c in dwg_decode ../../src/decode.c:226
    #5 0x7fb6378a687a in dwg_read_file ../../src/dwg.c:193
    #6 0x4019b5 in test_SVG ../../examples/dwg2svg2.c:119
    #7 0x404809 in main ../../examples/dwg2svg2.c:505
    #8 0x7fb63706882f in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x2082f)
    #9 0x4017e8 in _start (/home/dungnguyen/PoCs/libredwg_b0b41f5/.libs/lt-dwg2svg2+0x4017e8)
~~~

strongcourage, 2019-07-28:
#### 12.  SBO in dwg.c:1175 (#126-6)
PoC: https://github.com/strongcourage/PoCs/blob/master/libredwg_b0b41f5/PoC_sbo_dwg.c:1175
ASAN says:
~~~
==19015==ERROR: AddressSanitizer: stack-buffer-overflow on address 0x7ffd8372b1b8 at pc 0x7f88b65b6e6a bp 0x7ffd8372b170 sp 0x7ffd8372b160
WRITE of size 1 at 0x7ffd8372b1b8 thread T0
    #0 0x7f88b65b6e69 in dwg_section_type ../../src/dwg.c:1175
    #1 0x7f88b684e749 in read_sections_map ../../src/decode_r2007.c:893
    #2 0x7f88b684e749 in read_r2007_meta_data ../../src/decode_r2007.c:1684
    #3 0x7f88b6831767 in decode_R2007 ../../src/decode.c:2483
    #4 0x7f88b6831767 in dwg_decode ../../src/decode.c:222
    #5 0x7f88b65b287a in dwg_read_file ../../src/dwg.c:193
    #6 0x4019b5 in test_SVG ../../examples/dwg2svg2.c:119
    #7 0x404809 in main ../../examples/dwg2svg2.c:505
    #8 0x7f88b5d7482f in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x2082f)
    #9 0x4017e8 in _start (/home/dungnguyen/PoCs/libredwg_b0b41f5/.libs/lt-dwg2svg2+0x4017e8)
~~~

strongcourage, 2019-07-28:
#### 13. SEGV in dwg2svg2.c:258
PoC: https://github.com/strongcourage/PoCs/blob/master/libredwg_b0b41f5/PoC_segv_dwg2svg2.c:261
ASAN says:
~~~
==19095==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000018 (pc 0x0000004031c3 bp 0x7ffd061fbd80 sp 0x7ffd061fbbc0 T0)
    #0 0x4031c2 in output_INSERT ../../examples/dwg2svg2.c:258
    #1 0x4036d8 in output_object ../../examples/dwg2svg2.c:289
    #2 0x403b1a in output_BLOCK_HEADER ../../examples/dwg2svg2.c:350
    #3 0x4040bd in output_SVG ../../examples/dwg2svg2.c:403
    #4 0x4019c8 in test_SVG ../../examples/dwg2svg2.c:121
    #5 0x404809 in main ../../examples/dwg2svg2.c:505
    #6 0x7fe94f02682f in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x2082f)
    #7 0x4017e8 in _start (/home/dungnguyen/PoCs/libredwg_b0b41f5/.libs/lt-dwg2svg2+0x4017e8)
~~~

strongcourage, 2019-07-28:
#### 14. SEGV in free.c:820
PoC: https://github.com/strongcourage/PoCs/blob/master/libredwg_b0b41f5/PoC_segv_free.c:820
ASAN says:
~~~
==19179==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000008 (pc 0x7f15d5c3ee69 bp 0x7fffd4b9f0b0 sp 0x7fffd4b9f060 T0)
    #0 0x7f15d5c3ee68 in dwg_free ../../src/free.c:820
    #1 0x4019d2 in test_SVG ../../examples/dwg2svg2.c:123
    #2 0x404809 in main ../../examples/dwg2svg2.c:505
    #3 0x7f15d50ce82f in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x2082f)
    #4 0x4017e8 in _start (/home/dungnguyen/PoCs/libredwg_b0b41f5/.libs/lt-dwg2svg2+0x4017e8)
~~~

rurban, 2019-07-28:
As of 9ca3760a we have now the hbo of part 10 BLOCK_CONTROL.block_headers, 
and a msan leak of part 14 (read_2004_compressed_section decode.c:1961:26)

strongcourage, 2019-07-30:
Thanks for your patches on master. The latest commit bc7d9e0 fixed almost bugs. When running the latest commit with old crashing inputs, there are still following bugs that you could be interested to fix.
#### 15. HBO in bits.c:1826
PoC: https://github.com/strongcourage/PoCs/blob/master/libredwg_bc7d9e0/PoC_hbo_bits.c:1826
~~~
==17965==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x62a00000567f at pc 0x7f866c5a5d87 bp 0x7ffe9d426d20 sp 0x7ffe9d426d10
READ of size 1 at 0x62a00000567f thread T0
    #0 0x7f866c5a5d86 in bit_calc_CRC ../../src/bits.c:1826
    #1 0x7f866c8231ab in decode_R13_R2000 ../../src/decode.c:1062
    #2 0x7f866c8231ab in dwg_decode ../../src/decode.c:225
    #3 0x7f866c59a97a in dwg_read_file ../../src/dwg.c:193
    #4 0x401c2e in test_SVG ../../examples/dwg2svg2.c:119
    #5 0x401c2e in main ../../examples/dwg2svg2.c:506
    #6 0x7f866bd5c82f in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x2082f)
    #7 0x402038 in _start (/home/dungnguyen/PoCs/libredwg_bc7d9e0/.libs/lt-dwg2svg2+0x402038)
~~~

#### 16. HBO in bits.c:294
PoC: https://github.com/strongcourage/PoCs/blob/master/libredwg_bc7d9e0/PoC_hbo_bits.c:294
~~~
==18087==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x62c00000f600 at pc 0x7f94e96cc468 bp 0x7ffcab733b70 sp 0x7ffcab733b60
READ of size 1 at 0x62c00000f600 thread T0
    #0 0x7f94e96cc467 in bit_read_RC ../../src/bits.c:294
    #1 0x7f94e96cd85f in bit_read_H ../../src/bits.c:1120
    #2 0x7f94e96de493 in dwg_decode_handleref_with_code ../../src/decode.c:3283
    #3 0x7f94e97f791b in dwg_decode_VPORT_CONTROL_private ../../src/dwg.spec:2848
    #4 0x7f94e9938d0c in dwg_decode_VPORT_CONTROL ../../src/dwg.spec:2839
    #5 0x7f94e9938d0c in dwg_decode_add_object ../../src/decode.c:4302
    #6 0x7f94e993cae3 in read_2004_section_handles ../../src/decode.c:2316
    #7 0x7f94e993cae3 in decode_R2004 ../../src/decode.c:2491
    #8 0x7f94e9944e74 in dwg_decode ../../src/decode.c:231
    #9 0x7f94e96c697a in dwg_read_file ../../src/dwg.c:193
    #10 0x401c2e in test_SVG ../../examples/dwg2svg2.c:119
    #11 0x401c2e in main ../../examples/dwg2svg2.c:506
    #12 0x7f94e8e8882f in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x2082f)
    #13 0x402038 in _start (/home/dungnguyen/PoCs/libredwg_bc7d9e0/.libs/lt-dwg2svg2+0x402038)
~~~ 

#### 17. HBO in decode.c:1668
PoC: https://github.com/strongcourage/PoCs/blob/master/libredwg_bc7d9e0/PoC_hbo_decode.c:1668
~~~
==18106==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x61900000b8f8 at pc 0x7f6f4f7d5fe4 bp 0x7fff288a1220 sp 0x7fff288a1210
READ of size 8 at 0x61900000b8f8 thread T0
    #0 0x7f6f4f7d5fe3 in read_R2004_section_map ../../src/decode.c:1668
    #1 0x7f6f4f7d5fe3 in decode_R2004 ../../src/decode.c:2451
    #2 0x7f6f4f7d6e74 in dwg_decode ../../src/decode.c:231
    #3 0x7f6f4f55897a in dwg_read_file ../../src/dwg.c:193
    #4 0x401c2e in test_SVG ../../examples/dwg2svg2.c:119
    #5 0x401c2e in main ../../examples/dwg2svg2.c:506
    #6 0x7f6f4ed1a82f in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x2082f)
    #7 0x402038 in _start (/home/dungnguyen/PoCs/libredwg_bc7d9e0/.libs/lt-dwg2svg2+0x402038)
~~~

#### 18. SEGV in bits.c:295
PoC: https://github.com/strongcourage/PoCs/blob/master/libredwg_bc7d9e0/PoC_segv_bits.c:295
~~~
==18130==ERROR: AddressSanitizer: SEGV on unknown address 0x7fcaf3a87bdd (pc 0x7fcaf23173a9 bp 0x7ffee9a73d20 sp 0x7ffee9a73d10 T0)
    #0 0x7fcaf23173a8 in bit_read_RC ../../src/bits.c:295
    #1 0x7fcaf231885f in bit_read_H ../../src/bits.c:1120
    #2 0x7fcaf2329493 in dwg_decode_handleref_with_code ../../src/decode.c:3283
    #3 0x7fcaf2488adb in dwg_decode_LAYOUT_private ../../src/dwg.spec:3899
    #4 0x7fcaf255becc in dwg_decode_LAYOUT ../../src/dwg.spec:3842
    #5 0x7fcaf2584bbe in dwg_decode_add_object ../../src/decode.c:4391
    #6 0x7fcaf2587ae3 in read_2004_section_handles ../../src/decode.c:2316
    #7 0x7fcaf2587ae3 in decode_R2004 ../../src/decode.c:2491
    #8 0x7fcaf258fe74 in dwg_decode ../../src/decode.c:231
    #9 0x7fcaf231197a in dwg_read_file ../../src/dwg.c:193
    #10 0x401c2e in test_SVG ../../examples/dwg2svg2.c:119
    #11 0x401c2e in main ../../examples/dwg2svg2.c:506
    #12 0x7fcaf1ad382f in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x2082f)
    #13 0x402038 in _start (/home/dungnguyen/PoCs/libredwg_bc7d9e0/.libs/lt-dwg2svg2+0x402038)
~~~

rurban, 2019-07-31:
16 and 18 not so far. This requires the gh85-common branch, with stricter hdl_dat sizes.
The others are on my radar, but I got a couple of hundreds more, which are not that interesting and realistic yet.
I'll rather finish the importers first.

closed 2019-08-01 by rurban