LibreDWG/libredwg #144: ERROR: bit_read_RC buffer overflow at 267997258

epakshay, 2019-09-18:
I am using command line tool like : 
dwg2dxf  data.dwg

I tried converting few dwg file few worked properly. But when i tried to convert a file which have multiple feautre i.e point, line and polygon, it gives an error like this:
Reading DWG file data.dwg
ERROR: bit_read_RC buffer overflow at 267997258
ERROR: bit_read_RC buffer overflow at 267997258
ERROR: bit_read_RC buffer overflow at 267997258
ERROR: bit_read_RC buffer overflow at 267997258
ERROR: Invalid Data Section Page Map type 0x0 != 0x4163003b
ERROR: Failed to decode file: data.dwg 0x100

The file was created to plot few GIS i.e location related points and the size was upto 170MB. 
Please help me out in this. 
Thanks.

rurban, 2019-09-18:
Which DWG version exactly? can you post the header and the failing section part of the log, best with dwg2dxf - v3, thanks. Best would be the zipped dwg, sent to my email. 
It could be something like GH #93

epakshay, 2019-09-23:
Hi Reini,
I tried with dwg2dxf - v3, but the command does not respond. 
Not able to attach the zipped file here due to size limitation.
Here is the attached file link of google drive. 

[https://drive.google.com/open?id=1INk71FysZ69HjLP4k3XypjO3eopxDAqq](url)

Thanks

rurban, 2019-09-23:
Looks like an address overflow to me. 0xfff5c6a is almost at the end of 32bit (RL) types.
Might need to extend a few types from RL to long or RLL.

rurban, 2019-09-23:
I'm afraid you need to use the TeighaFileConverter for now. There's something wrong with the spec when reading so many sections which overflow the dwg size.

epakshay, 2019-09-23:
Ok. Thanks for the reply. How can i use TeighaFileConverter? Does it provide command line tool. I am searching for the same, but didn't find this conversion code. Can you please share its link so that i can look into its documentation too. Thanks

rurban, 2019-09-23:
The tetcases in GH #132 (also r2013) do have the same problems.

rurban, 2019-09-23:
https://www.google.com/search?q=TeighaFileConverter finds enough. This tool exists for linux, mac, windows, and can also be used via cmdline. see e.g. dxf-roundtrip.sh

rurban, 2022-04-08:
I got now a good testcase where libdxfrw works, and libredwg fails with section gaps. 
Thanks @timoria21 for the CFI dwg.

timoria21, 2022-04-11:
Thanks, happy to hear that these files helped you to fix some bugs.