LibreDWG/libredwg #152: docs for python bindings

looooo, 2019-10-02:
Are there any docs for the python bindings available. I have no idea how to use libredwg and run into a crash imediatly:

```
>>> import LibreDWG
>>> LibreDWG.Dwg_Data()
<LibreDWG.Dwg_Data; proxy of <Swig Object of type '_dwg_struct *' at 0x7f1dbf13d9c0> >
>>> data = LibreDWG.Dwg_Data()
>>> data.swig/python detected a memory leak of type 'BITCODE_MS *', no destructor found.
swig/python detected a memory leak of type 'uint16_t *', no destructor found.
swig/python detected a memory leak of type 'BITCODE_MS *', no destructor found.
swig/python detected a memory leak of type 'BITCODE_MS *', no destructor found.
```

looooo, 2019-10-02:
I guess the error is to due use of swig 4.0. I remember having similar issues with pivy which were solved by this commit:
https://github.com/FreeCAD/pivy/commit/262bd9513af6f11d2a1596e7d6cb556164ead24c

looooo, 2019-10-02:
So the python interface should work the same way as the c-interface? If so are there examples how to use libredwg c-interface?

rurban, 2019-10-02:
> So the python interface should work the same way as the c-interface? If so
> are there examples how to use libredwg c-interface?
Yes, all programs and examples and unit tests. The docs do have also a bit.

looooo, 2019-10-04:
I guess it would be best to add some docstrings to the wrapped library. Currently not much information is given:
[image]