LibreDWG/libredwg #156: fonts and some erro.

yoyomule, 2019-10-15:
Hello, I tested version 0.9 and I have the following problems:
I have provided two dwg files, tmp1.dwg and tmp2.dwg.
The tmp1.dwg file contains the shx font. The text content is well preserved when exporting, but all the characters are replaced with truetype fonts. So after dwg is converted into svg, many style changes have occurred. Can it still be shx font style restore?
tmp2.dwg contains a small amount of drawing content, but after switching out to svg, it does not see any content. I tried to use dwg2dxf to convert to dxf. At this time, I get the following error message.
[image]
[libredwg test.zip](https://github.com/LibreDWG/libredwg/files/3729861/libredwg.test.zip)
Thanks.

rurban, 2019-10-18:
* fonts: we don't yet translate font-styles to svg font-families. currently there's only Verdana. But I could change that to use Courier for all non-ttf fonts, and Verdana for ttf fonts. There's no SIMPLEX or COMPLEX in svg imho.

* error messages. That's no problem. some objects cannot be read yet. svg only translates a tiny percentage of all entities also.

closed 2019-10-18 by rurban

yoyomule, 2019-10-28:
> * L
Thanks.
Is it possible to turn shx into a symbol instead of a text so that it can be restored. You can add a choice when exporting.

rurban, 2019-10-28:
Unfortunately not. I don['t know a way to convert a SHX into blocks of polylines, maybe there's one.

commit d1012c1f (2019-10-18): dwg2SVG: honor TTF font styles => Verdana or Courier.

Translate font-styles to svg font-families.
Use Courier for all non-ttf fonts, and Arial or Verdana for ttf fonts.
There's no SIMPLEX or COMPLEX in svg imho.

Closes #156