LibreDWG/libredwg #165: Missing some ARC entities when decode the dwg file.

yz55556666, 2019-10-28:
I use 0.9 version of libredwg to decode a R2000 version of DWG file and draw all entities in the Dwg_Data, and find that there are some ARCs missing.
Here is a comparison between the results of my program drawing and the results of commercial software display.
[image]
[image]

The test file is here:
[test.zip](https://github.com/LibreDWG/libredwg/files/3777962/test.zip)

rurban, 2019-10-28:
How did you use it? via dxf2dwg? or dwg2dxf? or dwg2SVG?
It could be that those ARC's where inside blocks, which was only fixed recently.

yz55556666, 2019-10-28:
First of all, I use AutoCAD to decompose the drawings into the most basic drawings， then I use the following code for decode。

```
BITCODE_BL i;
    for (i = 0; i < dwg->num_objects; i++)
    {
        Dwg_Object *obj = &dwg->object[i];
        if (obj->supertype == DWG_SUPERTYPE_UNKNOWN) // unknown
            continue;
        if (obj->type == DWG_SUPERTYPE_OBJECT)       // no entity
            continue;
        if (obj->tio.entity->entmode == 0)           // belongs to block
            continue;
        if (!obj->tio.entity->invisible)
        {
            decodeObj(&dwg->object[i]);
        }
    }

```

rurban, 2019-10-28:
I see. One of the missing ARC's in question has handle 0.3.13AC6.
Properly decoded:
```
Object number: 34004/84D4, Size: 66 [MS], Type: 17 [BS]
Add entity ARC Decode entity ARC
bitsize: 504 [RL] @3.2
handle: 0.3.13AC6 [H 5]
EED[0] size: 0 [BS] (end)
preview_exists: 0 [B 0]
entmode: 2 [BB 0]
num_reactors: 0 [BL 0]
nolinks: 1 [B 0]
color.index: 256 [CMC.BS 62]
ltype_scale: 1.000000 [BD 48]
ltype_flags: 0 [BB 0]
plotstyle_flags: 0 [BB 0]
invisible: 0 [BS 60]
linewt: 0x3 [RC 370]
--common_size: 90
center: (-443672.959310, 261139.333514, 0.000000) [3BD 10]
radius: 926.250000 [BD 40]
thickness: 0.000000 [BT 39]
start_angle: 3.141593 [BD 50]  180º
end_angle: 4.686745 [BD 51]  268.531º
xdicobjhandle: (3.0.0) abs:0 [H 360]
layer: (5.1.10) abs:10 [H 8] => LAYER 0
 padding: +8 MISSING
 object_map{13AC6} = 34004
crc: 81D2 [RSx]
 check_CRC 2111010-2111078 = 68: 81D2 == 81D2
```

yz55556666, 2019-10-28:
I can't find the two arcs you mentioned by the following code，but the other arcs which visable can.
```
bool DwgConverter::decodeObj(Dwg_Object *obj)
{
    switch (obj->type)
    {
         case DWG_TYPE_ARC:
               if((int) arc->center.x == 443672){
                        break;
                }
                 break;
   }
```

yz55556666, 2019-10-28:
I have found the problem. The missing ARC has a wrong center.x. It can't be negative.
[image]
[image]

yz55556666, 2019-10-28:
This is the whole drawing. Some entities are in the y-axis mirror position.
At that time, I was looking for reasons. I thought these entities were caused by extension. z = - 1. Now it seems that they are not.
[image]

yz55556666, 2019-10-28:
This LWPOLYLINE has the wrong Points.x. 
[image]

yz55556666, 2019-10-29:
Sorry, I've got it wrong. If extension. z = - 1, I need to mirror the parameter by the Y axis. I was confused by the parameters displayed in AutoCAD.

yz55556666, 2019-10-29:
Compared with the parameters displayed in AutoCAD, the value of center. X of some arc is indeed inverse, which is different from that of ellipse, and the value of ellipse can completely correspond. So I think there is still a problem with decode the ARC entities .

rurban, 2019-10-29:
I see now what happened. When you exploded the $DorLib2D$00000001 blocks with these ARC's, the mirrored blocks (x -800) got exploded to mirrored extrusion (0,0,-1), that's why the center.x is negative.
If do do an DXFOUT of this dwg you'll get the same negative center.x points.
```
  0
ARC
  5
13AC6
330
1F
100
AcDbEntity
  8
0
370
    13
100
AcDbCircle
 10
-443672.9593103369
 20
261139.3335142909
 30
0.0
 40
926.2500000000001
210
0.0
220
0.0
230
-1.0
100
AcDbArc
 50
180.0
 51
268.5307164161525
```

so this not a bug, just a very silly representation. 
But there's is a bug, I'm missing the extrusion [3BD 210] (0,0,-1) for these mirrored arc's.

But I cannot even flip it around, because the DXF must need to carry the same values.

rurban, 2019-10-29:
special extrusion (0,0,-1) needs documentation, that .x points need to flip the sign then when displaying them.

yz55556666, 2019-10-29:
I don't agree with "explod problem"，because AutoCAD display is positive. I've tested that using vectors to correct this negative value is not the right way. Sometimes Z is not exactly equal to 1 .
Some LWPolyLine have this problem too. 
If I use Z to correct Ellipse will get the wrong answer, because the center. X of an ellipse with a negative Z value is a positive value.
[image]

yz55556666, 2019-10-29:
[image]

rurban, 2019-10-29:
see e.g. https://stackoverflow.com/questions/40669117/dxf-files-circle-coordinates-changed-its-sign-in-text-data for similar problems with flipped OCS, i.e. extrusion: (0.000000, 0.000000, -1.000000)

yz55556666, 2019-10-29:
Thank you very much for your help.

rurban, 2019-10-29:
Note that an Ellipse is an exception to the OCS rule. See the DXF docs, e.g. https://buildmedia.readthedocs.org/media/pdf/ezdxf/latest/ezdxf.pdf

closed 2019-10-29 by rurban

rurban, 2019-10-29:
I've added now geometric utilities to help in displaying OCS entities, used by dwg2SVG and dwg2ps

yz55556666, 2019-10-30:
Lines can'not use the "transform_OCS", when z=0, the resault is wrong. Some Lines has z=0 value.
The resault will be very different when z=-1 and  z = -1.0000000000002, and cause the wrong result.
[image]

frankg1, 2022-04-20:
大佬您好 arc的情况确实如您所说，但是circle的圆心左边基本上都是乱的，跟本来的坐标完全不搭噶。所以想请教一下你，能不能给点建议出来呢？

commit 5790568a (2019-10-30): add OCS doc
Closes #165
commit 2e6f99a4 (2019-10-30): add OCS doc
Closes #165