LibreDWG/libredwg #166: Odd buffer output

fpmalard, 2019-11-04:
Hi,

I'm trying to run the dwglayers.exe app with some DWG files.
It works just fine when called via BAT command but when called from a .NET app using Process class it works sometimes and fails to get information from some dwg files.

`
ProcessStartInfo objStartInfo = new ProcessStartInfo();
            objStartInfo.FileName = "dwglayers.exe";
            objStartInfo.Arguments = string.Format("\"{0}\"", "MyDwgFile.dwg");
            objStartInfo.RedirectStandardOutput = true;
            objStartInfo.UseShellExecute = false;
            objStartInfo.CreateNoWindow = true;

            try
            {
                List<string> lstResult = new List<string>();
                using (var objProcess = new Process())
                {
                    objProcess.EnableRaisingEvents = true;
                    objProcess.StartInfo = objStartInfo;
                    objProcess.OutputDataReceived += (sender, e) =>
                    {
                        if (string.IsNullOrWhiteSpace(e.Data))
                            return;

                        if (e.Data.Trim() == "0")
                            return;

                        if (!lstResult.Any(sItem => sItem.Equals(e.Data, StringComparison.InvariantCultureIgnoreCase)))
                            lstResult.Add(e.Data);
                    };

                    objProcess.Start();
                    objProcess.BeginOutputReadLine();

                    objProcess.WaitForExit();
                }
            }
            catch (Exception excError)
            {
            }
`

It seems dwglayers.exe doesn't fill the output before exiting (or maybe doesn't flush the output properly).

The odd behavior is that for some drawings this code works just fine.
I'm attaching a DWG that can be successfully run via BAT but can't with this code.

It should list the following layers:
0
AFUWS-P
AFUWS-P-TN

Thank you.
[MyDwgFile.dwg.zip](https://github.com/LibreDWG/libredwg/files/3804033/MyDwgFile.dwg.zip)

fpmalard, 2019-11-04:
Maybe there is an exception happening at line 172:
`dwg_free (&dwg);`

For this specific DWG and it is not finishing the stdout.
Maybe a call to:
`fflush(stdout);`

after finishing the FOR loop to assure the buffer will be flush at that time.

Any thoughts?

rurban, 2019-11-05:
The exception happened no in dwg_free, but earlier, in the loop already. 
There may be NULL layer_control.entries.
But a flush makes sense after every line, as the loop is too fragile.

closed 2019-11-05 by rurban

fpmalard, 2019-11-05:
> The exception happened no in dwg_free, but earlier, in the loop already.
> There may be NULL layer_control.entries.
> But a flush makes sense after every line, as the loop is too fragile.
> 
> I think that requires now another patch release 0.9.2.

Would you be able to provide a new `dwglayers.exe` binary along `libredwg-0.dll`  with the latest release version?
I can test my .NET code again and check if it solved the issue.

Thank you.

rurban, 2019-11-05:
New Window binaries are built automatically with every change to master: https://github.com/LibreDWG/libredwg/releases =>
i.e. https://github.com/LibreDWG/libredwg/releases/tag/0.9.1.2450

fpmalard, 2019-11-05:
Got the updated version but when I try to run it requires a dll called: `libssp-0.dll`
The release version I had didn't require that.

fpmalard, 2019-11-05:
I see, but will this dll be distributed with the Windows binary or do I need to get it from somewhere else?