LibreDWG/libredwg #167: Layers with spaces in their names are being renamed with "_" chair replacing these spaces

fpmalard, 2019-11-05:
It seems layer names in the format "ABCD(space)ABDC" are being renamed to "ABCD_ABCD" where any space char is replaced by a "_" char.

Sample file attached and screenshot inside AutoCAD comparing to the dwglayers.exe output.
It seems the issue doesn't have to do with dwglayers app itself.

Maybe something with this code (dwg_api.c):
`bit_convert_TU ((BITCODE_TU)layer->name);`

https://github.com/LibreDWG/libredwg/blob/a09c1996ee226de18c6de5d1f7b7a26996a75440/src/dwg_api.c#L20140

[chair.zip](https://github.com/LibreDWG/libredwg/files/3809770/chair.zip)
[image]

rurban, 2019-11-05:
Interesting, I'll check.
bit_convert_TU just converts UTF-2 to utf-8 display or DXF, but doesn't mangle any space characters.

rurban, 2019-11-06:
The DWG format stores the layer name with the underscore, i.e. "ARM_CAPS", and the application displays some of those names with a space instead, i.e. "ARM CAPS". I guess they do it for backwards compatibility with older 3rdparty apps, which cannot handle space in table names.
The conversion of those names is stored in a XRECORD with EXTNAMES.
In our case:
```
XRECORD
handle: 0.1.EA [H 5]
xdata[0]: "EXTNAMES" [TF 8 102]
xdata[1]: "ARM_CAPS" [TF 8 1]
xdata[2]: "ARM CAPS" [TF 8 2]
...
```
which is owned by a DICTIONARY named ACAD_XREC_ROUNDTRIP, which is owned by the LAYER ARM_CAPS.

so dwglayer could be extended (by some new flag) to display the layers EXTNAMES, with the space instead of the underscore.

fpmalard, 2019-11-06:
> The DWG format stores the layer name with the underscore, i.e. "ARM_CAPS", and the application displays some of those names with a space instead, i.e. "ARM CAPS". I guess they do it for backwards compatibility with older 3rdparty apps and acad 14, which cannot handle space in table names. see the systemvariable EXTNAMES.
> 
> The conversion of those names is stored in a XRECORD with EXTNAMES.
> In our case:
> 
> ```
> XRECORD
> handle: 0.1.EA [H 5]
> xdata[0]: "EXTNAMES" [TF 8 102]
> xdata[1]: "ARM_CAPS" [TF 8 1]
> xdata[2]: "ARM CAPS" [TF 8 2]
> ...
> ```
> 
> which is owned by a DICTIONARY named ACAD_XREC_ROUNDTRIP, which is owned by the LAYER ARM_CAPS.
> 
> so dwglayer could be extended (by some new flag) to display the layers EXTNAMES, with the space instead of the underscore.

Yes, indeed. `EXTNAMES` was created to support old drawing table name limitations.
Thank you for addressing this and look forward to get the `dwglayers.exe` updated version.

closed 2019-11-07 by rurban