LibreDWG/libredwg #170: Could not build libredwg in Ubuntu 18.04.2 LTS

Vishnu-C, 2019-11-28:
On make i get these error
`make[2]: Entering directory '/media/vishnu/LocalDisk1/SourceCodes/libredwg/doc'
/usr/bin/perl ../src/gen-dynapi.pl
Can't locate Convert/Binary/C.pm in @INC (you may need to install the Convert::Binary::C module) (@INC contains: /etc/perl /usr/local/lib/x86_64-linux-gnu/perl/5.26.1 /usr/local/share/perl/5.26.1 /usr/lib/x86_64-linux-gnu/perl5/5.26 /usr/share/perl5 /usr/lib/x86_64-linux-gnu/perl/5.26 /usr/share/perl/5.26 /usr/local/lib/site_perl /usr/lib/x86_64-linux-gnu/perl-base) at ../src/gen-dynapi.pl line 23.
BEGIN failed--compilation aborted at ../src/gen-dynapi.pl line 23.
Makefile:902: recipe for target 'dynapi.texi' failed
make[2]: [dynapi.texi] Error 2 (ignored)
  MAKEINFO LibreDWG.info
/media/vishnu/LocalDisk1/SourceCodes/libredwg/build-aux/missing: line 81: makeinfo: command not found
WARNING: 'makeinfo' is missing on your system.
         You should only need it if you modified a '.texi' file, or
         any other file indirectly affecting the aspect of the manual.
         You might want to install the Texinfo package:
         <http://www.gnu.org/software/texinfo/>
         The spurious makeinfo call might also be the consequence of
         using a buggy 'make' (AIX, DU, IRIX), in which case you might
         want to install GNU make:
         <http://www.gnu.org/software/make/>
Makefile:447: recipe for target 'LibreDWG.info' failed
make[2]: *** [LibreDWG.info] Error 127
make[2]: Leaving directory '/media/vishnu/LocalDisk1/SourceCodes/libredwg/doc'
Makefile:594: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/media/vishnu/LocalDisk1/SourceCodes/libredwg'
Makefile:489: recipe for target 'all' failed
make: *** [all] Error 2
`

rurban, 2019-11-29:
The first problem can be ignored. Or install this missing perl module.
The 2nd problem is only when making the documentation via make info, which is missing. The library and programs should have been compiled successfully.

Vishnu-C, 2019-11-30:
Thanks for the reply.
How can i generate a make file to run a example? I don't see a cmakeLists to generate build scripts.

Vishnu-C, 2019-12-01:
sample camke file if someone needs it

```
cmake_minimum_required(VERSION 2.8.2)

project(loadDWG)

set(NAMER_DIR ${CMAKE_CURRENT_SOURCE_DIR})

set( LIBREDWG_INCLUDE_PATH path/libredwg/include CACHE
    PATH "libredwg headers location" )
set( LIBREDWG_SRC_PATH path/libredwg/src CACHE
    PATH "libredwg src location" )
set( LIBREDWG_PROG_PATH path/libredwg/programs CACHE
    PATH "libredwg program location" )
set( LIBREDWG_LIB_PATH path/libredwg/src/.libs CACHE
    PATH "libredwg libraries location" )
set(LIBREDWG_LIBRARIES libredwg.so)
# Tells CMake where are the required headers
include_directories(${LIBREDWG_INCLUDE_PATH} ${LIBREDWG_SRC_PATH} ${LIBREDWG_PROG_PATH})
LINK_DIRECTORIES(${LIBREDWG_LIB_PATH})

# namer
include_directories(${NAMER_DIR})

file(GLOB NAMER_srcs 
    "${NAMER_DIR}/dwgrewrite.c" 
    )

add_executable(loadDWG ${NAMER_srcs})
target_link_libraries(loadDWG ${LIBREDWG_LIBRARIES})
```

closed 2019-12-01 by Vishnu-C