LibreDWG/libredwg #171: dwgrewrite.c results in errors

Vishnu-C, 2019-12-01:
dwgrewrite.c present in programs reads a dwg file and rewrites. 
using `dwg_write_file` results errors and the export file cannot be opened.

`Reading DWG file ../libredwg/test/test-data/sample_2000.dwg
ERROR: Second header num_handlers > 16: 17259

Writing DWG file sample_2000-rewrite.dwg
Warning: Wrong object size: 18240 + 182 = 18422 != 18431: -9 off
Warning: Wrong object size: 18446 + 30 = 18476 != 18477: -1 off
Warning: Wrong object size: 18498 + 22 = 18520 != 18521: -1 off
Warning: Wrong object size: 18524 + 37 = 18561 != 18565: -4 off
Warning: Wrong object size: 18565 + 19 = 18584 != 18585: -1 off
Warning: Wrong object size: 18588 + 279 = 18867 != 18870: -3 off
Warning: Wrong object size: 18871 + 279 = 19150 != 19153: -3 off
Warning: Wrong object size: 19154 + 293 = 19447 != 19449: -2 off
Warning: Wrong object size: 19451 + 29 = 19480 != 19481: -1 off
Warning: Wrong object size: 19484 + 62 = 19546 != 19549: -3 off
Warning: Wrong object size: 19571 + 46 = 19617 != 19620: -3 off
Warning: Wrong object size: 19621 + 39 = 19660 != 19665: -5 off
Warning: Wrong object size: 19664 + 29 = 19693 != 19694: -1 off
Warning: Wrong object size: 19717 + 181 = 19898 != 19907: -9 off
Warning: Wrong object size: 19902 + 118 = 20020 != 20025: -5 off
Warning: Wrong object size: 20024 + 37 = 20061 != 20066: -5 off
Warning: Wrong object size: 20065 + 85 = 20150 != 20160: -10 off
Warning: Wrong object size: 20154 + 29 = 20183 != 20184: -1 off
Warning: Wrong object size: 20207 + 37 = 20244 != 20249: -5 off
Warning: Wrong object size: 20248 + 85 = 20333 != 20343: -10 off
Warning: Wrong object size: 20337 + 32 = 20369 != 20370: -1 off
Warning: Wrong object size: 20395 + 54 = 20449 != 20452: -3 off
Warning: Wrong object size: 20453 + 17 = 20470 != 20471: -1 off
Warning: Wrong object size: 20474 + 18 = 20492 != 20493: -1 off
Warning: Wrong object size: 20552 + 29 = 20581 != 20582: -1 off
Warning: Wrong object size: 20585 + 118 = 20703 != 20708: -5 off
Warning: Wrong object size: 20752 + 86 = 20838 != 20839: -1 off
Warning: Wrong object size: 21053 + 154 = 21207 != 21208: -1 off
Warning: Wrong object size: 21211 + 45 = 21256 != 21258: -2 off
Warning: Wrong object size: 21321 + 37 = 21358 != 21359: -1 off
Warning: Wrong object size: 21453 + 37 = 21490 != 21491: -1 off
Warning: Wrong object size: 21585 + 37 = 21622 != 21623: -1 off
Warning: Wrong object size: 21717 + 16 = 21733 != 21735: -2 off

Re-reading created file sample_2000-rewrite.dwg
ERROR: bit_read_RC buffer overflow at 182
ERROR: bit_read_RC buffer overflow at 182
ERROR: bit_read_RC buffer overflow at 182
ERROR: bit_read_RC buffer overflow at 182
ERROR: bit_read_RC buffer overflow at 182
ERROR: bit_read_RC buffer overflow at 30
ERROR: bit_read_RC buffer overflow at 37
ERROR: bit_read_RC buffer overflow at 37
ERROR: bit_read_RC buffer overflow at 37
ERROR: bit_read_RC buffer overflow at 19
ERROR: bit_read_RC buffer overflow at 29
ERROR: bit_read_RC buffer overflow at 62
ERROR: bit_read_RC buffer overflow at 62
ERROR: bit_read_RC buffer overflow at 39
ERROR: bit_read_RC buffer overflow at 39
ERROR: bit_read_RC buffer overflow at 39
ERROR: bit_read_RC buffer overflow at 29
ERROR: bit_read_RC buffer overflow at 181
ERROR: bit_read_RC buffer overflow at 181
ERROR: bit_read_RC buffer overflow at 181
ERROR: bit_read_RC buffer overflow at 181
ERROR: bit_read_RC buffer overflow at 181
ERROR: bit_read_RC buffer overflow at 181
ERROR: bit_read_RC buffer overflow at 118
ERROR: bit_read_RC buffer overflow at 118
ERROR: bit_read_RC buffer overflow at 118
ERROR: bit_read_RC buffer overflow at 118
ERROR: bit_read_RC buffer overflow at 118
ERROR: bit_read_RC buffer overflow at 37
ERROR: bit_read_RC buffer overflow at 37
ERROR: bit_read_RC buffer overflow at 37
ERROR: bit_read_RC buffer overflow at 85
ERROR: bit_read_RC buffer overflow at 85
ERROR: bit_read_RC buffer overflow at 85
ERROR: bit_read_RC buffer overflow at 85
ERROR: bit_read_RC buffer overflow at 85
ERROR: bit_read_RC buffer overflow at 85
ERROR: bit_read_RC buffer overflow at 85
ERROR: bit_read_RC buffer overflow at 29
ERROR: bit_read_RC buffer overflow at 37
ERROR: bit_read_RC buffer overflow at 37
ERROR: bit_read_RC buffer overflow at 37
ERROR: bit_read_RC buffer overflow at 85
ERROR: bit_read_RC buffer overflow at 85
ERROR: bit_read_RC buffer overflow at 85
ERROR: bit_read_RC buffer overflow at 85
ERROR: bit_read_RC buffer overflow at 85
ERROR: bit_read_RC buffer overflow at 85
ERROR: bit_read_RC buffer overflow at 85
ERROR: bit_read_RC buffer overflow at 32
Warning: Invalid handle-reference, longer than 8 bytes: 5.14.0
Warning: Invalid handle-reference, longer than 8 bytes: 5.14.0
ERROR: bit_read_RC buffer overflow at 118
ERROR: bit_read_RC buffer overflow at 118
ERROR: bit_read_RC buffer overflow at 118
ERROR: bit_read_RC buffer overflow at 118
ERROR: bit_read_RC buffer overflow at 118
ERROR: bit_read_RC buffer overflow at 86
ERROR: bit_read_RC buffer overflow at 154
ERROR: bit_read_RC buffer overflow at 45
ERROR: bit_read_RC buffer overflow at 37
ERROR: bit_read_RC buffer overflow at 37
ERROR: bit_read_RC buffer overflow at 37
`

rurban, 2019-12-01:
Yes, that's why we have it there.

Vishnu-C, 2019-12-01:
So it is not yet complete?

rurban, 2019-12-01:
No

Vishnu-C, 2019-12-01:
Sorry what did you mean by no? is it complete?

rurban, 2019-12-01:
No

closed 2019-12-26 by rurban