LibreDWG/libredwg #175: Docker image make check fails on import libxml2

lynochka, 2019-12-20:
I am trying to build libredwg docker image
based on the following dockerfile: 

```
FROM debian:sid
RUN echo 'debconf debconf/frontend select Noninteractive' | debconf-set-selections
RUN sed -i 's/archive/ru.archive/g' /etc/apt/sources.list \
 && apt-get update                                        \
 && apt-get install -y git make gcc autoconf libtool      \
        pkg-config libxml2-dev dejagnu python-dev swig    \
	zlib1g-dev texinfo                                \
 && apt-get clean                                         \
 && rm -rf /var/lib/{apt,dpkg,cache,log}/

RUN git clone git://git.sv.gnu.org/libredwg.git
```
)

In the interactive container I try to execute the following:
```
sh ./autogen.sh
./configure
make 
make check 
```

Which gives:
Warnings during `make`:
```
bits.c:1160:2: warning: #warning x86_64 gcc-9.2 codegen is seriously broken! better use an older version [-Wcpp]
 1160 | #warning x86_64 gcc-9.2 codegen is seriously broken! better use an older version
```
```
/usr/include/x86_64-linux-gnu/bits/string_fortified.h:106:10: warning: '__builtin_strncpy' output may be truncated copying 70 bytes from a string of length 79 [-Wstringop-truncation]
  106 |   return __builtin___strncpy_chk (__dest, __src, __len, __bos (__dest));
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

Errors during `make check`:
```
make  check-TESTS
make[4]: Entering directory '/home/libredwg/test/xmlsuite'
Traceback (most recent call last):
  File "./check.py", line 8, in <module>
    from helper import *
  File "/home/libredwg/test/xmlsuite/helper.py", line 2, in <module>
    import libxml2
ImportError: No module named libxml2
FAIL: check.py
```

Warnings during `make check`:
```
tests_common.h:28:1: warning: 'ms_printf' is an unrecognized format function type [-Wformat=]
   28 | static void fail (const char *fmt, ...) ATTRIBUTE_FORMAT (1, 2);
tests_common.h:52:3: warning: function 'ok' might be a candidate for 'gnu_printf' format attribute [-Wsuggest-attribute=format]
   52 |   vsnprintf (buffer, sizeof (buffer), fmt, ap);
tests_common.h: In function 'fail':
tests_common.h:72:3: warning: function 'fail' might be a candidate for 'gnu_printf' format attribute [-Wsuggest-attribute=format]
   72 |   vsnprintf (buffer, sizeof (buffer), fmt, ap);
In file included from hash_test.c:6:
tests_common.h:28:1: warning: 'ms_printf' is an unrecognized format function type [-Wformat=]
   28 | static void fail (const char *fmt, ...) ATTRIBUTE_FORMAT (1, 2);
```

Could you please help to make a good docker image where I could test dwg to svg functionality and figure what is wrong?

rurban, 2019-12-20:
debian:sid needs also `python-libxml2` for the tests. that's your problem.
configure should really check for that.

on debian gcc-9.2 is really broken, but I recently had better experiences on fedora 31 gcc-9.2, don't know what redhat had fixed there. on mingw64 it's also broken.
maybe an alpine, void or fedora docker is better?

The ms_printf warnings are my fault, will be fixed. but it's harmless.
The new strncpy truncation warning seems to be harmless, I'll check.

lynochka, 2019-12-20:
Could you please share with me a dockerfile setup that works for you?

lynochka, 2019-12-20:
For dwg2SVG I am mostly looking into extracting the texts (and its positions)

rurban, 2019-12-25:
Sorry, I don't have docker. The CI instances use docker, look at `.travis.yml` or the other yml's for a clean setup with all reqs.

closed 2019-12-25 by rurban