LibreDWG/libredwg #178: Several bugs need to be fixed.

wu610777031, 2019-12-31:
Hi,
I got some bugs, and I tested on master branch and version **0.9.3**. There are **3 heap overflow**, **2 NULL pointer deference**, **1 denial of service**, **1 stack overflow** (this bug causes memory leak in master branch) in that.
Compile with **ASAN** and use **degrewrite** to repro that.
Hear are some details:

wu610777031, 2019-12-31:
**file:heap_overflow1**
**1. Crafted input will lead to heap-buffer-overflow in /src/bits.c:1370 **
ASAN report:
```
==97792==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x60300000e1b5 at pc 0x7ff0165d944e bp 0x7ffc703987e0 sp 0x7ffc703987d0
READ of size 1 at 0x60300000e1b5 thread T0
    #0 0x7ff0165d944d in bit_write_TF /home/sv/Documents/libredwg-master/src/bits.c:1370
    #1 0x7ff0173a08d3 in dwg_encode_eed_data /home/sv/Documents/libredwg-master/src/encode.c:2069
    #2 0x7ff0173a1e49 in dwg_encode_eed /home/sv/Documents/libredwg-master/src/encode.c:2178
    #3 0x7ff0173a2ccf in dwg_encode_entity /home/sv/Documents/libredwg-master/src/encode.c:2257
    #4 0x7ff0172751f0 in dwg_encode_MTEXT /home/sv/Documents/libredwg-master/src/dwg.spec:2225
    #5 0x7ff01739ea00 in dwg_encode_add_object /home/sv/Documents/libredwg-master/src/encode.c:1808
    #6 0x7ff01721b341 in dwg_encode /home/sv/Documents/libredwg-master/src/encode.c:1191
    #7 0x7ff0165c72ec in dwg_write_file /home/sv/Documents/libredwg-master/src/dwg.c:350
    #8 0x555c7778444e in main /home/sv/Documents/libredwg-master/programs/dwgrewrite.c:286
    #9 0x7ff015cdab96 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x21b96)
    #10 0x555c77783149 in _start (/home/sv/output3/bin/dwgrewrite+0x2149)

0x60300000e1b5 is located 0 bytes to the right of 21-byte region [0x60300000e1a0,0x60300000e1b5)
allocated by thread T0 here:
    #0 0x7ff0179eed38 in __interceptor_calloc (/usr/lib/x86_64-linux-gnu/libasan.so.4+0xded38)
    #1 0x7ff0166665a2 in dwg_decode_eed /home/sv/Documents/libredwg-master/src/decode.c:3242
    #2 0x7ff016668ebf in dwg_decode_entity /home/sv/Documents/libredwg-master/src/decode.c:3550
    #3 0x7ff0167f7c56 in dwg_decode_MTEXT_private /home/sv/Documents/libredwg-master/src/dwg.spec:2225
    #4 0x7ff0167f78fa in dwg_decode_MTEXT /home/sv/Documents/libredwg-master/src/dwg.spec:2225
    #5 0x7ff016a778ca in dwg_decode_add_object /home/sv/Documents/libredwg-master/src/decode.c:4799
    #6 0x7ff01663bc5d in decode_R13_R2000 /home/sv/Documents/libredwg-master/src/decode.c:1240
    #7 0x7ff0165ddb7e in dwg_decode /home/sv/Documents/libredwg-master/src/decode.c:239
    #8 0x7ff0165c5a65 in dwg_read_file /home/sv/Documents/libredwg-master/src/dwg.c:206
    #9 0x555c77783ffb in main /home/sv/Documents/libredwg-master/programs/dwgrewrite.c:226
    #10 0x7ff015cdab96 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x21b96)

SUMMARY: AddressSanitizer: heap-buffer-overflow /home/sv/Documents/libredwg-master/src/bits.c:1370 in bit_write_TF
Shadow bytes around the buggy address:
  0x0c067fff9be0: 00 00 00 00 fa fa 00 00 00 00 fa fa 00 00 00 00
  0x0c067fff9bf0: fa fa 00 00 00 00 fa fa 00 00 00 00 fa fa 00 00
  0x0c067fff9c00: 00 00 fa fa 00 00 00 00 fa fa 00 00 00 00 fa fa
  0x0c067fff9c10: 00 00 00 00 fa fa 00 00 00 00 fa fa 00 00 00 00
  0x0c067fff9c20: fa fa 00 00 00 00 fa fa 00 00 00 00 fa fa 00 00
=>0x0c067fff9c30: 00 01 fa fa 00 00[05]fa fa fa 00 00 00 00 fa fa
  0x0c067fff9c40: 00 00 00 00 fa fa 00 00 00 00 fa fa 00 00 00 00
  0x0c067fff9c50: fa fa 00 00 00 00 fa fa 00 00 00 00 fa fa 00 00
  0x0c067fff9c60: 00 00 fa fa 00 00 00 00 fa fa 00 00 00 00 fa fa
  0x0c067fff9c70: 00 00 00 00 fa fa 00 00 00 00 fa fa 00 00 03 fa
  0x0c067fff9c80: fa fa 00 00 00 00 fa fa 00 00 00 00 fa fa 00 00
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
```

wu610777031, 2019-12-31:
**file:heap_overflow2
1. Crafted input will lead to heap-buffer-overflow in src/common_entity_data.spec:175**
ASAN report:
```
==97787==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x608000008f18 at pc 0x7f98cbbefd57 bp 0x7ffc08008710 sp 0x7ffc08008700
READ of size 8 at 0x608000008f18 thread T0
    #0 0x7f98cbbefd56 in dwg_encode_entity /home/sv/Documents/libredwg-master/src/common_entity_data.spec:175
    #1 0x7f98cbbb9197 in dwg_encode_UNKNOWN_ENT /home/sv/Documents/libredwg-master/src/dwg.spec:5928
    #2 0x7f98cbbea541 in dwg_encode_add_object /home/sv/Documents/libredwg-master/src/encode.c:1964
    #3 0x7f98cba66341 in dwg_encode /home/sv/Documents/libredwg-master/src/encode.c:1191
    #4 0x7f98cae122ec in dwg_write_file /home/sv/Documents/libredwg-master/src/dwg.c:350
    #5 0x55cfb2b8144e in main /home/sv/Documents/libredwg-master/programs/dwgrewrite.c:286
    #6 0x7f98ca525b96 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x21b96)
    #7 0x55cfb2b80149 in _start (/home/sv/output3/bin/dwgrewrite+0x2149)

0x608000008f18 is located 8 bytes to the left of 88-byte region [0x608000008f20,0x608000008f78)
allocated by thread T0 here:
    #0 0x7f98cc239d38 in __interceptor_calloc (/usr/lib/x86_64-linux-gnu/libasan.so.4+0xded38)
    #1 0x7f98cb268af4 in dwg_add_UNKNOWN_OBJ /home/sv/Documents/libredwg-master/src/dwg.spec:5934
    #2 0x7f98cb268f0e in dwg_decode_UNKNOWN_OBJ /home/sv/Documents/libredwg-master/src/dwg.spec:5934
    #3 0x7f98cb2c4aac in dwg_decode_add_object /home/sv/Documents/libredwg-master/src/decode.c:5019
    #4 0x7f98cae86c5d in decode_R13_R2000 /home/sv/Documents/libredwg-master/src/decode.c:1240
    #5 0x7f98cae28b7e in dwg_decode /home/sv/Documents/libredwg-master/src/decode.c:239
    #6 0x7f98cae10a65 in dwg_read_file /home/sv/Documents/libredwg-master/src/dwg.c:206
    #7 0x55cfb2b80ffb in main /home/sv/Documents/libredwg-master/programs/dwgrewrite.c:226
    #8 0x7f98ca525b96 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x21b96)

SUMMARY: AddressSanitizer: heap-buffer-overflow /home/sv/Documents/libredwg-master/src/common_entity_data.spec:175 in dwg_encode_entity
Shadow bytes around the buggy address:
  0x0c107fff9190: fa fa fa fa 00 00 00 00 00 00 00 00 00 00 00 fa
  0x0c107fff91a0: fa fa fa fa 00 00 00 00 00 00 00 00 00 00 00 fa
  0x0c107fff91b0: fa fa fa fa 00 00 00 00 00 00 00 00 00 00 00 fa
  0x0c107fff91c0: fa fa fa fa 00 00 00 00 00 00 00 00 00 00 00 fa
  0x0c107fff91d0: fa fa fa fa 00 00 00 00 00 00 00 00 00 00 00 fa
=>0x0c107fff91e0: fa fa fa[fa]00 00 00 00 00 00 00 00 00 00 00 fa
  0x0c107fff91f0: fa fa fa fa 00 00 00 00 00 00 00 00 00 00 00 fa
  0x0c107fff9200: fa fa fa fa 00 00 00 00 00 00 00 00 00 00 00 fa
  0x0c107fff9210: fa fa fa fa 00 00 00 00 00 00 00 00 00 00 00 fa
  0x0c107fff9220: fa fa fa fa 00 00 00 00 00 00 00 00 00 00 00 fa
  0x0c107fff9230: fa fa fa fa 00 00 00 00 00 00 00 00 00 00 00 fa
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
```

wu610777031, 2019-12-31:
**file:heap_overflow3**
**3. Crafted input will lead to heap-buffer-overflow in src/decode.c:1339**
ASAN report:
```
==97993==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x6130000001b0 at pc 0x7f8fbb2298e5 bp 0x7ffec24076a0 sp 0x7ffec2407690
READ of size 8 at 0x6130000001b0 thread T0
    #0 0x7f8fbb2298e4 in decode_R13_R2000 /home/sv/Documents/libredwg-master/src/decode.c:1339
    #1 0x7f8fbb1cab7e in dwg_decode /home/sv/Documents/libredwg-master/src/decode.c:239
    #2 0x7f8fbb1b2a65 in dwg_read_file /home/sv/Documents/libredwg-master/src/dwg.c:206
    #3 0x55edd3fc9ffb in main /home/sv/Documents/libredwg-master/programs/dwgrewrite.c:226
    #4 0x7f8fba8c7b96 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x21b96)
    #5 0x55edd3fc9149 in _start (/home/sv/output3/bin/dwgrewrite+0x2149)

0x6130000001b0 is located 8 bytes to the right of 360-byte region [0x613000000040,0x6130000001a8)
allocated by thread T0 here:
    #0 0x7f8fbc5dbd38 in __interceptor_calloc (/usr/lib/x86_64-linux-gnu/libasan.so.4+0xded38)
    #1 0x7f8fbb219d8f in decode_R13_R2000 /home/sv/Documents/libredwg-master/src/decode.c:885
    #2 0x7f8fbb1cab7e in dwg_decode /home/sv/Documents/libredwg-master/src/decode.c:239
    #3 0x7f8fbb1b2a65 in dwg_read_file /home/sv/Documents/libredwg-master/src/dwg.c:206
    #4 0x55edd3fc9ffb in main /home/sv/Documents/libredwg-master/programs/dwgrewrite.c:226
    #5 0x7f8fba8c7b96 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x21b96)

SUMMARY: AddressSanitizer: heap-buffer-overflow /home/sv/Documents/libredwg-master/src/decode.c:1339 in decode_R13_R2000
Shadow bytes around the buggy address:
  0x0c267fff7fe0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c267fff7ff0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c267fff8000: fa fa fa fa fa fa fa fa 00 00 00 00 00 00 00 00
  0x0c267fff8010: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c267fff8020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
=>0x0c267fff8030: 00 00 00 00 00 fa[fa]fa fa fa fa fa fa fa fa fa
  0x0c267fff8040: fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd
  0x0c267fff8050: fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd
  0x0c267fff8060: fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd
  0x0c267fff8070: fa fa fa fa fa fa fa fa fd fd fd fd fd fd fd fd
  0x0c267fff8080: fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
```

wu610777031, 2019-12-31:
**file:null_pointer1**
**4. There is a NULL pointer dereference in the function /src/dwg.spec:4117**
ASAN report
```
==97830==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000000 (pc 0x7fd08188a801 bp 0x7ffdac74f740 sp 0x7ffdac74f6e0 T0)
==97830==The signal is caused by a READ memory access.
==97830==Hint: address points to the zero page.
    #0 0x7fd08188a800 in dwg_encode_LWPOLYLINE /home/sv/Documents/libredwg-master/src/dwg.spec:4117
    #1 0x7fd081916d87 in dwg_encode_add_object /home/sv/Documents/libredwg-master/src/encode.c:1897
    #2 0x7fd081793341 in dwg_encode /home/sv/Documents/libredwg-master/src/encode.c:1191
    #3 0x7fd080b3f2ec in dwg_write_file /home/sv/Documents/libredwg-master/src/dwg.c:350
    #4 0x55dc68f1944e in main /home/sv/Documents/libredwg-master/programs/dwgrewrite.c:286
    #5 0x7fd080252b96 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x21b96)
    #6 0x55dc68f18149 in _start (/home/sv/output3/bin/dwgrewrite+0x2149)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV /home/sv/Documents/libredwg-master/src/dwg.spec:4117 in dwg_encode_LWPOLYLINE
```

wu610777031, 2019-12-31:
**file:null_pointer2**
**5. There is a NULL pointer dereference in the function /src/common_entity_handle_data.spec:37**
ASAN report:
```
==97720==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000000 (pc 0x7f91db437390 bp 0x7ffed659dc00 sp 0x7ffed659d7c0 T0)
==97720==The signal is caused by a READ memory access.
==97720==Hint: address points to the zero page.
    #0 0x7f91db43738f in dwg_encode_common_entity_handle_data /home/sv/Documents/libredwg-master/src/common_entity_handle_data.spec:37
    #1 0x7f91db3a4957 in dwg_encode_LWPOLYLINE /home/sv/Documents/libredwg-master/src/dwg.spec:4132
    #2 0x7f91db42fd87 in dwg_encode_add_object /home/sv/Documents/libredwg-master/src/encode.c:1897
    #3 0x7f91db2ac341 in dwg_encode /home/sv/Documents/libredwg-master/src/encode.c:1191
    #4 0x7f91da6582ec in dwg_write_file /home/sv/Documents/libredwg-master/src/dwg.c:350
    #5 0x5632dd18b44e in main /home/sv/Documents/libredwg-master/programs/dwgrewrite.c:286
    #6 0x7f91d9d6bb96 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x21b96)
    #7 0x5632dd18a149 in _start (/home/sv/output3/bin/dwgrewrite+0x2149)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV /home/sv/Documents/libredwg-master/src/common_entity_handle_data.spec:37 in dwg_encode_common_entity_handle_data
```

wu610777031, 2019-12-31:
**file:dos**
**6. Crafted input will lead to denial of service in /src/bits.c:2069, for loop.**
Stack overview:
```
pwndbg> bt
#0  bit_calc_CRC (len=913972745, addr=0x7fff76407000 "", seed=<optimized out>) at bits.c:2071
#1  bit_write_CRC (dat=dat@entry=0x7fffffffd300, start_address=start_address@entry=1692740325, seed=seed@entry=49345) at bits.c:1312
#2  0x00007ffff7b15fb5 in dwg_encode_add_object (obj=0x5555557cfd20, dat=0x7fffffffd300, address=1692740325) at encode.c:2049
#3  0x00007ffff7b1e6f6 in dwg_encode (dwg=0x7fffffffd4c0, dat=dat@entry=0x7fffffffd300) at encode.c:1191
#4  0x00007ffff77e8034 in dwg_write_file (filename=0x5555557582c0 "findings:id:000153,sig:11,src:000000,op:flip1,pos:398050-rewrite.dwg", dwg=<optimized out>) at dwg.c:350
#5  0x00005555555551e7 in main (argc=argc@entry=2, argv=argv@entry=0x7fffffffe558) at dwgrewrite.c:286
#6  0x00007ffff6fb5b97 in __libc_start_main (main=0x555555554e20 <main>, argc=2, argv=0x7fffffffe558, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffe548) at ../csu/libc-start.c:310
#7  0x000055555555548a in _start ()
```

wu610777031, 2019-12-31:
**file:memory_leak**
**6. Crafted input will lead to stack overflow in 0.9.3 or memory leak in master branch in /src/bits.c:1355.**
ASAN report:
```
==97764==ERROR: LeakSanitizer: detected memory leaks

Direct leak of 25 byte(s) in 1 object(s) allocated from:
    #0 0x7f8aba412b50 in __interceptor_malloc (/usr/lib/x86_64-linux-gnu/libasan.so.4+0xdeb50)
    #1 0x7f8ab8ffd38a in bit_read_TF /home/sv/Documents/libredwg-master/src/bits.c:1355
    #2 0x7f8ab908a0e6 in dwg_decode_eed /home/sv/Documents/libredwg-master/src/decode.c:3232
    #3 0x7f8ab908cebf in dwg_decode_entity /home/sv/Documents/libredwg-master/src/decode.c:3550
    #4 0x7f8ab937ad6f in dwg_decode_LWPOLYLINE_private /home/sv/Documents/libredwg-master/src/dwg.spec:4048
    #5 0x7f8ab937aa13 in dwg_decode_LWPOLYLINE /home/sv/Documents/libredwg-master/src/dwg.spec:4048
    #6 0x7f8ab949d2fa in dwg_decode_add_object /home/sv/Documents/libredwg-master/src/decode.c:4945
    #7 0x7f8ab905fc5d in decode_R13_R2000 /home/sv/Documents/libredwg-master/src/decode.c:1240
    #8 0x7f8ab9001b7e in dwg_decode /home/sv/Documents/libredwg-master/src/decode.c:239
    #9 0x7f8ab8fe9a65 in dwg_read_file /home/sv/Documents/libredwg-master/src/dwg.c:206
    #10 0x5588b8500531 in main /home/sv/Documents/libredwg-master/programs/dwgrewrite.c:303
    #11 0x7f8ab86feb96 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x21b96)

SUMMARY: AddressSanitizer: 25 byte(s) leaked in 1 allocation(s).
```

wu610777031, 2019-12-31:
[poc.zip](https://github.com/LibreDWG/libredwg/files/4012028/poc.zip)
pocs are here.

rurban, 2019-12-31:
I haven't fuzzed `dwgrewrite` yet, thanks. Just started fuzzing dxf2dwg only.

wu610777031, 2019-12-31:
> I haven't fuzzed `dwgrewrite` yet, thanks. Just started fuzzing dxf2dwg only.

No worries, many people are fuzzing ```libredwg``` as far as i know.

rurban, 2019-12-31:
I'm having about 170 crashes in my pipeline, which seem to be ~4 missing new protections

wu610777031, 2019-12-31:
> I'm having about 170 crashes in my pipeline, which seem to be ~4 missing new protections

Yeah, I got ~700 crashes before, but I think these 7 bugs are useful. And GOOD job, these bugs seem to be done.

wu610777031, 2019-12-31:
By the way, could you please request some cves for me? And if I can get CVE-2020-xxxx?

rurban, 2020-01-01:
head_overflow2 is the most interesting and only remaining one.
The TABLE 0.2.9BC is not detected as entity in decode.

wu610777031, 2020-01-02:
> head_overflow2 is the most interesting and only remaining one.
> The TABLECONTENT 0.2.9BC is wrongly encoded as TABLE entity, mixed up. (but only here in a fuzzed DXF, not in reality)
> 
> ```
> Class 528 0x401 ACAD_TABLE
>  AcDbTable "ObjectDBX Classes" 0 0x1f2
> Class 529 0x480 TABLE
>  AcDbTableContent "ObjectDBX Classes" 0 0x0
> 
> < Object number: 468/1D4, Size: 8316 [MS], Type: 529 [BS]
> Warning: Unknown Class object 529 TABLE (0x480)
> Add object UNKNOWN_OBJ Decode object UNKNOWN_OBJ
> 
> > Object number: 468, Size: 8316 [MS], Type: 529 [BS]
> Warning: Unhandled Class entity 528 ACAD_TABLE (0x401) 468/9BC
> Encode entity UNKNOWN_ENT
> ```

Exactly! This issue made me confused. So weired the structure is:
```
  number = 506, 
  proxyflag = 1153, 
  appname = 0x6030000009d0 "ObjectDBX Classes", 
  cppname = 0x6020000001f0 "AcDbMaterial", 
  dxfname = 0x602000000210 "MATERIAL", 
  dxfname_u = 0x0, 
  wasazombie = 0 '\000', 
  item_class_id = 499, 
  num_instances = 0, 
  dwg_version = 0, 
  maint_version = 0, 
  unknown_1 = 0, 
  unknown_2 = 0
```

rurban, 2020-01-02:
All fixed in master since 3f503dd294efc63a59608d8a16058c41d44ba13a

closed 2020-01-02 by rurban

wu610777031, 2020-01-02:
> All fixed in master since [3f503dd](https://github.com/LibreDWG/libredwg/commit/3f503dd294efc63a59608d8a16058c41d44ba13a)

Well down! Are you requesting CVEs for me?

wu610777031, 2020-01-02:
These bugs are credited by Skyvast in ADLab of Venustech.

rurban, 2020-01-02:
No CVE's by me. Too busy preparing the next release.

ShangzhiXu, 2025-01-06:
FYI,
**CVE-2019-20915** is assigned for heap_overflow1
**CVE-2019-20913** for heap_overflow2
**CVE-2019-20910** for heap_overflow3