LibreDWG/libredwg #186: NULL pointer deference

wu610777031, 2020-01-14:
Hello, I got a NULL pointer deference bug by run `dxf2dwg poc -o /dev/null`
```
==12391==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000000 (pc 0x7f9f9b4f3284 bp 0x7ffcfa3b08a0 sp 0x7ffcfa3b0860 T0)
==12391==The signal is caused by a READ memory access.
==12391==Hint: address points to the zero page.
    #0 0x7f9f9b4f3283 in add_MLINESTYLE_lines /home/skyvast/Documents/libredwg-0.10.1.2677/src/in_dxf.c:1462
    #1 0x7f9f9b554a23 in new_object /home/skyvast/Documents/libredwg-0.10.1.2677/src/in_dxf.c:5897
    #2 0x7f9f9b564d2d in dxf_objects_read /home/skyvast/Documents/libredwg-0.10.1.2677/src/in_dxf.c:7245
    #3 0x7f9f9b56bb16 in dwg_read_dxf /home/skyvast/Documents/libredwg-0.10.1.2677/src/in_dxf.c:7701
    #4 0x7f9f9a684ee7 in dxf_read_file /home/skyvast/Documents/libredwg-0.10.1.2677/src/dwg.c:319
    #5 0x564a81d20465 in main /home/skyvast/Documents/libredwg-0.10.1.2677/programs/dxf2dwg.c:255
    #6 0x7f9f99d8fb96 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x21b96)
    #7 0x564a81d1f489 in _start (/home/skyvast/Documents/asan_libredwg/bin/dxf2dwg+0x2489)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV /home/skyvast/Documents/libredwg-0.10.1.2677/src/in_dxf.c:1462 in add_MLINESTYLE_lines
==12391==ABORTING

```

wu610777031, 2020-01-14:
[image]
remove`.jpg`

rurban, 2020-01-14:
Can you attach the poc please?

wu610777031, 2020-01-14:
[null_pointer.zip](https://github.com/LibreDWG/libredwg/files/4059197/null_pointer.zip)Can you reproduce that?

wu610777031, 2020-01-14:
> Can you attach the poc please?

Can you reproduce it?
[null_pointer.zip](https://github.com/LibreDWG/libredwg/files/4059214/null_pointer.zip)

wu610777031, 2020-01-15:
[null_pointer.zip](https://github.com/LibreDWG/libredwg/files/4064145/null_pointer.zip)

closed 2020-01-16 by rurban

wu610777031, 2020-07-20:
This bug credited by ADLab. 
CVE-2020-15807