LibreDWG/libredwg #195: dwg export with freecad

looooo, 2020-01-28:
FreeCAD now allows to export dwg's with libredwg if libredwg is available. We cannot make libredwg a hard dependency because of license issues (lgpl2 vs gpl3).

We use the dwg2dxf command for importing dwgs and dxf2dwg for the export. The import seems to work for most objects. But the export has bigger issues:

```
Warning: dxf2dwg is still highly experimental.
Reading DXF file /tmp/tmp6_fiede5/test.dxf
Warning: Misleading LTYPE_CONTROL.num_entries 0 for 0th entry
Warning: Misleading LTYPE_CONTROL.num_entries 0 for 0th entry
Warning: Misleading LAYER_CONTROL.num_entries 3 for 0th entry
Warning: Misleading LAYER_CONTROL.num_entries 4 for 1th entry
Warning: Misleading LAYER_CONTROL.num_entries 5 for 2th entry
Warning: Misleading STYLE_CONTROL.num_entries 2 for 0th entry
Warning: Misleading STYLE_CONTROL.num_entries 3 for 1th entry
Warning: Misleading APPID_CONTROL.num_entries 2 for 0th entry
Warning: Misleading APPID_CONTROL.num_entries 3 for 1th entry
Warning: Misleading DIMSTYLE_CONTROL.num_entries 2 for 0th entry
Warning: Unknown DXF code 287 for DIMSTYLE
Warning: Misleading BLOCK_CONTROL.num_entries 5 for 0th entry
Warning: Misleading BLOCK_CONTROL.num_entries 5 for 0th entry
Warning: Unknown HEADER.DIMSTYLE STANDARD dxf:3
Warning: Unknown HEADER.CMLSTYLE STANDARD dxf:2
ERROR: Unexpected DXF end-of-file
Warning: Unknown HEADER.DIMSTYLE STANDARD dxf:3
Warning: Unknown HEADER.CMLSTYLE STANDARD dxf:2
Warning: Object handle not found, 1/1 in 33 objects
Warning: Object handle not found, 12/C in 33 objects
Writing DWG file /home/lo/Schreibtisch/test.dwg
```

So regarding this message dxf2dwg is not ready to use yet. Would be nice to get some feedback how much is missing for a working dxf2dwg and if there is anything testers/users can do to accelerate this process?

some version info
```
libredwg                  0.10.1.2729
OS: Ubuntu 18.04.3 LTS (ubuntu:GNOME/ubuntu)
```

rurban, 2020-01-28:
Great! Exciting.

Those warning are bogus and can be ignored, I'll remove them.
```
Warning: Misleading LTYPE_CONTROL.num_entries 0 for 0th entry
Warning: Misleading LTYPE_CONTROL.num_entries 0 for 0th entry
Warning: Misleading LAYER_CONTROL.num_entries 3 for 0th entry
Warning: Misleading LAYER_CONTROL.num_entries 4 for 1th entry
Warning: Misleading LAYER_CONTROL.num_entries 5 for 2th entry
Warning: Misleading STYLE_CONTROL.num_entries 2 for 0th entry
Warning: Misleading STYLE_CONTROL.num_entries 3 for 1th entry
Warning: Misleading APPID_CONTROL.num_entries 2 for 0th entry
Warning: Misleading APPID_CONTROL.num_entries 3 for 1th entry
Warning: Misleading DIMSTYLE_CONTROL.num_entries 2 for 0th entry
Warning: Misleading BLOCK_CONTROL.num_entries 5 for 0th entry
Warning: Misleading BLOCK_CONTROL.num_entries 5 for 0th entry
```
It was just a reminder during development of the importer, and I forgot to remove them. DXF has different structures than DWG.

Those errors need to be investigated:
```
Warning: Unknown DXF code 287 for DIMSTYLE
Warning: Unknown HEADER.DIMSTYLE STANDARD dxf:3
Warning: Unknown HEADER.CMLSTYLE STANDARD dxf:2
Warning: Object handle not found, 1/1 in 33 objects
Warning: Object handle not found, 12/C in 33 objects
```

looooo, 2020-02-29:
thanks for working on this issue and sry for not reporting earlier. I tested with libredwg 
and see this output for the export:

```
Warning: dxf2dwg is still highly experimental.
Reading DXF file /tmp/tmp0s3hle72/test1.dxf
Warning: Misleading LTYPE_CONTROL.num_entries 0 for 0th entry
Warning: Misleading LTYPE_CONTROL.num_entries 0 for 0th entry
Warning: Misleading LAYER_CONTROL.num_entries 3 for 0th entry
Warning: Misleading LAYER_CONTROL.num_entries 4 for 1th entry
Warning: Misleading LAYER_CONTROL.num_entries 5 for 2th entry
Warning: Misleading STYLE_CONTROL.num_entries 2 for 0th entry
Warning: Misleading STYLE_CONTROL.num_entries 3 for 1th entry
Warning: Misleading APPID_CONTROL.num_entries 2 for 0th entry
Warning: Misleading APPID_CONTROL.num_entries 3 for 1th entry
Warning: Misleading DIMSTYLE_CONTROL.num_entries 2 for 0th entry
Warning: Unknown DXF code 287 for DIMSTYLE
Warning: Misleading BLOCK_CONTROL.num_entries 5 for 0th entry
Warning: Misleading BLOCK_CONTROL.num_entries 5 for 0th entry
Warning: Unknown HEADER.DIMSTYLE STANDARD dxf:3
Warning: Unknown HEADER.CMLSTYLE STANDARD dxf:2
ERROR: Unexpected DXF end-of-file
Warning: Unknown HEADER.DIMSTYLE STANDARD dxf:3
Warning: Unknown HEADER.CMLSTYLE STANDARD dxf:2
Warning: Object handle not found, 1/1 in 31 objects
Warning: Object handle not found, 12/C in 31 objects
Writing DWG file /home/lo/Schreibtisch/test1.dwg
```

I tested to open the exported dwg with qcad and librecad, but it doesn't seem to work.

looooo, 2020-03-10:
Basic export import test is still not working with v0.10.1.2986:

writing:
```
Warning: dxf2dwg is still highly experimental.
Reading DXF file /tmp/tmp85nv60zf/test_1.dxf
Warning: Unknown DXF code 287 for DIMSTYLE
Warning: Unknown HEADER.DIMSTYLE STANDARD dxf:3
Warning: Unknown HEADER.CMLSTYLE STANDARD dxf:2
ERROR: Unexpected DXF end-of-file
Warning: Unknown HEADER.DIMSTYLE STANDARD dxf:3
Warning: Unknown HEADER.CMLSTYLE STANDARD dxf:2
Warning: Object handle not found, 1/1 in 31 objects
Warning: Object handle not found, 12/C in 31 objects
Writing DWG file /home/lo/Schreibtisch/test_1.dwg

```

reading:
```
Reading DWG file /home/lo/Schreibtisch/test_1.dwg
Warning: Invalid handle-reference, longer than 8 bytes: 4.9.0
Warning: Invalid handle-reference, longer than 8 bytes: 10.10.0
Warning: Invalid handle-reference, longer than 8 bytes: 10.10.0
Warning: Invalid handle-reference, longer than 8 bytes: 10.10.0
ERROR: bit_read_BD: unexpected 2-bit code: '11'
ERROR: Invalid 3BD EXTMIN
Warning: Ignore invalid handleoff (@2533)
ERROR: Invalid object handle 0.0.0 at pos @6.2
ERROR: bit_read_TV buffer overflow at 91, length 256
ERROR: bit_read_RC buffer overflow at 119
ERROR: bit_read_RC buffer overflow at 119
ERROR: bit_read_RC buffer overflow at 119
ERROR: bit_read_RC buffer overflow at 119
ERROR: bit_read_RC buffer overflow at 119
ERROR: bit_read_RC buffer overflow at 119
ERROR: bit_read_RC buffer overflow at 119
ERROR: bit_advance_position buffer overflow at pos 118, size 119, advance by 1
ERROR: bit_read_RC buffer overflow at 119
ERROR: bit_advance_position buffer overflow at pos 118, size 119, advance by 2
ERROR: bit_read_RC buffer overflow at 119
ERROR: bit_read_RC buffer overflow at 119
ERROR: bit_read_RC buffer overflow at 119
ERROR: bit_read_RC buffer overflow at 119
ERROR: bit_read_RC buffer overflow at 119
ERROR: bit_read_RC buffer overflow at 119
ERROR: bit_read_RC buffer overflow at 119
ERROR: bit_read_RC buffer overflow at 119
ERROR: bit_advance_position buffer overflow at pos 118, size 119, advance by 2
ERROR: bit_read_BD: unexpected 2-bit code: '11'
ERROR: Invalid BD DIMTSZ
Warning: Ignore invalid handleoff (@2574)
Warning: Object handle not found, 1/1 in 31 objects
Writing DXF file /tmp/tmpn1h07oxu/test_1.dxf
ERROR: BLOCK_HEADER.block_entity missing

```

rurban, 2020-04-11:
I've made all Unknown DXF codes an hard error on all stable/supported objects.
Unstable objects are not imported to DWG, but replaced by a DUMMY placeholder.

Unstable objects are still tried to be parsed tough, so some of those warnings may still appear. Esp. the overflows, which will appear on parsing errors. There might be even more, as now the parser buffer is restricted to a single object to avoid overflows.
You'll care mostly about entities (graphical objects), where only a few are remaining unstable.
Roundtrips are now hardened and tested ok.

I've added a better internal interchange format as JSON, which can be tested much easier. 
The importer is very general, not so many special cases as with DXF.
With this format and roundtrips I could fix many remaining encoder errors.

3DSOLID's ACIS entities and MESH/SURFACE are still problematic, esp. with the new AcDs data storage and the Binary SAT format. Older formats do like fine.

I'm not so sure about SPLINE's. Looks like they are undertested, and the DXF format is very confusing. The two scenarios (rational B-Spline and Bezier, i.e. fit pts or not) are being mixed up there.

looooo, 2020-04-15:
Thanks @rurban for working on this. We see some progress with basic dwg export/import. Basic types are now working. I hope for someone showing up and further optimizing the freecad side of the dxf/dwg exporter. I guess at one point we might want to use the python bindings of libredwg, although I was not able to build the bindings for windows. That's another topic to tackle in the next days. If you have any thoughts on that, please leave a note [here](https://github.com/conda-forge/libredwg-feedstock/pull/37).

rurban, 2020-04-15:
Just merged many dxf/dwg fixes into master. Only minor ones remain.

looooo, 2020-06-26:
Hi @rurban 

we currently see some python errors:

```
ImportError: /home/conda/feedstock_root/build_artifacts/libredwg_1592928860281/_test_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placeho/lib/python3.6/site-packages/_LibreDWG.so: undefined symbol: dwg_setup_PERSUBENTMGR
```

maybe you can have a look.
https://dev.azure.com/conda-forge/feedstock-builds/_build/results?buildId=177575&view=results

rurban, 2020-07-11:
Some indxf (dxf2dwg - DXF import) limitaions do remain, but freecad and normal command-line usage should be okay now for most cases.

closed 2020-07-11 by rurban