LibreDWG/libredwg #200: Getting error while trying to convert dwg to json

vimalpillai21, 2020-02-06:
I am getting the following error while converting to json. 

dwgread -O JSON -o jw.json JW.dwg
Warning: Unhandled Class object 501 MATERIAL (0x481) 14/0
Warning: Unhandled Class object 501 MATERIAL (0x481) 22/0
Warning: Unhandled Class object 501 MATERIAL (0x481) 30/0
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 39/0
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 40/0
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 41/0
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 42/0
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 43/0
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 44/0
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 45/0
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 46/0
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 47/0
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 48/0
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 49/0
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 50/0
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 51/0
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 52/0
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 53/0
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 54/0
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 55/0
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 56/0
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 57/0
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 58/0
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 59/0
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 60/0
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 61/0
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 62/0
Warning: Unstable Class object 504 TABLESTYLE (0xfff) 95/0
ERROR: TABLESTYLE r2010+ not yet implemented
Warning: Unhandled Class object 506 ACDBSECTIONVIEWSTYLE (0x401) 99/0
Warning: Unhandled Class object 507 ACDBDETAILVIEWSTYLE (0x401) 101/0
Warning: Unknown Class object 509 ACMESCOPE (0x400)
Warning: Unknown Class object 510 ACMECOMMANDHISTORY (0x400)
Warning: Unknown Class object 511 ACIMSYSVARMAN (0x400)
Warning: Unknown Class object 512 ACIMINTSYSVAR (0x400)
Warning: Unknown Class object 513 ACMESTATEMGR (0x400)
Warning: Unknown Class object 515 ACGREFMASTER (0x400)
Warning: Unknown Class object 518 ACGREFACADMASTER (0x400)
Warning: Unknown Class object 521 AMIMASTER (0x400)
Warning: Unknown Class object 522 AMGSETTINGSOBJ (0x400)
Warning: Unknown Class object 523 AMGOBJPROPCFG (0x400)
Warning: Unknown Class object 522 AMGSETTINGSOBJ (0x400)
Warning: Unknown Class object 524 MAPIAMMOUDLE (0x400)
Warning: Unknown Class object 511 ACIMSYSVARMAN (0x400)
Warning: Unknown Class object 525 ACMSTANDARDANSI (0x400)
Warning: Unknown Class object 526 ACMSURFSTANDARDANSI (0x400)
Warning: Unknown Class object 527 ACMFCFSTANDARDANSI (0x400)
Warning: Unknown Class object 528 ACMWELDSTANDARDANSI (0x400)
Warning: Unknown Class object 529 ACMIDSTANDARDANSI (0x400)
Warning: Unknown Class object 530 ACMDATUMSTANDARDANSI (0x400)
Warning: Unknown Class object 531 ACMBOMSTANDARDANSI (0x400)
Warning: Unknown Class object 532 ACMDETAILSTANDARDANSI (0x400)
Warning: Unknown Class object 533 ACMSECTIONSTANDARDANSI (0x400)
Warning: Unknown Class object 534 ACMCENTERLINESTANDARDANSI (0x400)
Warning: Unknown Class object 535 ACMTHREADLINESTANDARDANSI (0x400)
Warning: Unknown Class object 536 ACAMGLAYSTD (0x400)
Warning: Unknown Class object 537 ACAMGDIMSTD (0x400)
Warning: Unknown Class object 538 ACAMGTITLESTD (0x400)
Warning: Unknown Class object 539 ACMNOTESTANDARDANSI (0x400)
Warning: Unknown Class object 540 ACMTAPERSTANDARDANSI (0x400)
Warning: Unknown Class object 541 ACMANOOTATIONVIEWSTANDARDANSI (0x400)
Warning: Unknown Class object 542 ACMSTANDARDDIN (0x400)
Warning: Unknown Class object 543 AMDTEDGESTANDARDDIN (0x400)
Warning: Unknown Class object 544 ACMSURFSTANDARDDIN (0x400)
Warning: Unknown Class object 545 ACMFCFSTANDARDDIN (0x400)
Warning: Unknown Class object 546 ACMWELDSTANDARDDIN (0x400)
Warning: Unknown Class object 547 ACMIDSTANDARDDIN (0x400)
[... 202 lines cut ...]
Warning: Unknown Class entity 609 SURFACE (0xfff)
Warning: Unknown Class entity 609 SURFACE (0xfff)
Warning: Unknown Class entity 609 SURFACE (0xfff)
Warning: Unknown Class entity 609 SURFACE (0xfff)
Warning: Unknown Class entity 609 SURFACE (0xfff)
Warning: Unknown Class entity 609 SURFACE (0xfff)
Warning: Unknown Class entity 609 SURFACE (0xfff)
Warning: Unknown Class entity 609 SURFACE (0xfff)
Warning: Unknown Class entity 609 SURFACE (0xfff)
Warning: Unknown Class entity 609 SURFACE (0xfff)
Warning: Unknown Class entity 609 SURFACE (0xfff)
Warning: Unknown Class entity 609 SURFACE (0xfff)
Warning: Unknown Class entity 609 SURFACE (0xfff)
Warning: Unknown Class entity 609 SURFACE (0xfff)
Warning: Unknown Class entity 609 SURFACE (0xfff)
Warning: Unknown Class entity 609 SURFACE (0xfff)
Warning: Unhandled Class object 610 RAPIDRTRENDERSETTINGS (0x400) 1119/0
Warning: Unhandled Class object 610 RAPIDRTRENDERSETTINGS (0x400) 1120/0
Warning: Unhandled Class object 610 RAPIDRTRENDERSETTINGS (0x400) 1121/0
Warning: Unhandled Class object 610 RAPIDRTRENDERSETTINGS (0x400) 1122/0
Warning: Unhandled Class object 610 RAPIDRTRENDERSETTINGS (0x400) 1123/0
Warning: Unhandled Class object 610 RAPIDRTRENDERSETTINGS (0x400) 1124/0
Warning: Unhandled Class object 610 RAPIDRTRENDERSETTINGS (0x400) 1125/0
Warning: thumbnail.size mismatch: 5261 != 0
SUCCESS

rurban, 2020-02-06:
Nice, some new classes I haven't seen before. Thanks.

rurban, 2020-02-07:
See #201, same root cause and proper explanation in the error and warning messages.
Some objects are ignored, some unstable ones are tried but fail. Still better than teigha in some cases.

closed 2020-02-07 by rurban

vimalpillai21, 2020-02-11:
I have tried again to convert the same file from dwg to json. Could you please tell what shall I do? Then I encountered this problem:

```
Warning: Unhandled Class object 501 MATERIAL (0x481) 14/0
Warning: Unhandled Class object 501 MATERIAL (0x481) 22/0
Warning: Unhandled Class object 501 MATERIAL (0x481) 30/0
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 39/0
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 40/0
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 41/0
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 42/0
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 43/0
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 44/0
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 45/0
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 46/0
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 47/0
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 48/0
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 49/0
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 50/0
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 51/0
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 52/0
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 53/0
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 54/0
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 55/0
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 56/0
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 57/0
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 58/0
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 59/0
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 60/0
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 61/0
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 62/0
Warning: Unstable Class object 504 TABLESTYLE (0xfff) 95/0
ERROR: TABLESTYLE r2010+ not yet implemented
Warning: Unhandled Class object 506 ACDBSECTIONVIEWSTYLE (0x401) 99/0
Warning: Unhandled Class object 507 ACDBDETAILVIEWSTYLE (0x401) 101/0
Warning: Unknown Class object 509 ACMESCOPE (0x400)
Warning: Unknown Class object 510 ACMECOMMANDHISTORY (0x400)
Warning: Unknown Class object 511 ACIMSYSVARMAN (0x400)
Warning: Unknown Class object 512 ACIMINTSYSVAR (0x400)
Warning: Unknown Class object 513 ACMESTATEMGR (0x400)
Warning: Unknown Class object 515 ACGREFMASTER (0x400)
Warning: Unknown Class object 518 ACGREFACADMASTER (0x400)
Warning: Unknown Class object 521 AMIMASTER (0x400)
Warning: Unknown Class object 522 AMGSETTINGSOBJ (0x400)
Warning: Unknown Class object 523 AMGOBJPROPCFG (0x400)
Warning: Unknown Class object 522 AMGSETTINGSOBJ (0x400)
Warning: Unknown Class object 524 MAPIAMMOUDLE (0x400)
Warning: Unknown Class object 511 ACIMSYSVARMAN (0x400)
Warning: Unknown Class object 525 ACMSTANDARDANSI (0x400)
Warning: Unknown Class object 526 ACMSURFSTANDARDANSI (0x400)
Warning: Unknown Class object 527 ACMFCFSTANDARDANSI (0x400)
Warning: Unknown Class object 528 ACMWELDSTANDARDANSI (0x400)
Warning: Unknown Class object 529 ACMIDSTANDARDANSI (0x400)
Warning: Unknown Class object 530 ACMDATUMSTANDARDANSI (0x400)
Warning: Unknown Class object 531 ACMBOMSTANDARDANSI (0x400)
Warning: Unknown Class object 532 ACMDETAILSTANDARDANSI (0x400)
Warning: Unknown Class object 533 ACMSECTIONSTANDARDANSI (0x400)
Warning: Unknown Class object 534 ACMCENTERLINESTANDARDANSI (0x400)
Warning: Unknown Class object 535 ACMTHREADLINESTANDARDANSI (0x400)
Warning: Unknown Class object 536 ACAMGLAYSTD (0x400)
Warning: Unknown Class object 537 ACAMGDIMSTD (0x400)
Warning: Unknown Class object 538 ACAMGTITLESTD (0x400)
Warning: Unknown Class object 539 ACMNOTESTANDARDANSI (0x400)
Warning: Unknown Class object 540 ACMTAPERSTANDARDANSI (0x400)
Warning: Unknown Class object 541 ACMANOOTATIONVIEWSTANDARDANSI (0x400)
Warning: Unknown Class object 542 ACMSTANDARDDIN (0x400)
Warning: Unknown Class object 543 AMDTEDGESTANDARDDIN (0x400)
Warning: Unknown Class object 544 ACMSURFSTANDARDDIN (0x400)
Warning: Unknown Class object 545 ACMFCFSTANDARDDIN (0x400)
Warning: Unknown Class object 546 ACMWELDSTANDARDDIN (0x400)
Warning: Unknown Class object 547 ACMIDSTANDARDDIN (0x400)
[... 202 lines cut ...]
Warning: Unknown Class entity 609 SURFACE (0xfff)
Warning: Unknown Class entity 609 SURFACE (0xfff)
Warning: Unknown Class entity 609 SURFACE (0xfff)
Warning: Unknown Class entity 609 SURFACE (0xfff)
Warning: Unknown Class entity 609 SURFACE (0xfff)
Warning: Unknown Class entity 609 SURFACE (0xfff)
Warning: Unknown Class entity 609 SURFACE (0xfff)
Warning: Unknown Class entity 609 SURFACE (0xfff)
Warning: Unknown Class entity 609 SURFACE (0xfff)
Warning: Unknown Class entity 609 SURFACE (0xfff)
Warning: Unknown Class entity 609 SURFACE (0xfff)
Warning: Unknown Class entity 609 SURFACE (0xfff)
Warning: Unknown Class entity 609 SURFACE (0xfff)
Warning: Unknown Class entity 609 SURFACE (0xfff)
Warning: Unknown Class entity 609 SURFACE (0xfff)
Warning: Unknown Class entity 609 SURFACE (0xfff)
Warning: Unhandled Class object 610 RAPIDRTRENDERSETTINGS (0x400) 1119/0
Warning: Unhandled Class object 610 RAPIDRTRENDERSETTINGS (0x400) 1120/0
Warning: Unhandled Class object 610 RAPIDRTRENDERSETTINGS (0x400) 1121/0
Warning: Unhandled Class object 610 RAPIDRTRENDERSETTINGS (0x400) 1122/0
Warning: Unhandled Class object 610 RAPIDRTRENDERSETTINGS (0x400) 1123/0
Warning: Unhandled Class object 610 RAPIDRTRENDERSETTINGS (0x400) 1124/0
Warning: Unhandled Class object 610 RAPIDRTRENDERSETTINGS (0x400) 1125/0
Warning: thumbnail.size mismatch: 5261 != 0
```

rurban, 2020-02-11:
What can you do?
You can either try adding support for these objects, or ignore it.

I think you want to create a 3D renderer. So the SURFACE support is lacking. I believe that's a B-Rep, but simply represented in the 3DSOLID ACIS format.
configure with `CFLAGS=-DDEBUG_CLASSES`, and then you get experimental parsing support for some unhandled classes.
Then implement ACIS support. It's a pretty simple textual format.