LibreDWG/libredwg #201: Getting error while trying to convert from dwg to dxf file

vimalpillai21, 2020-02-06:
I am getting this error while trying to convert dwg file to dxf format. Could you please check I have attached the dwg file below?
[JW.zip](https://github.com/LibreDWG/libredwg/files/4163532/JW.zip)

```
vimal-pillai@blackbird:~/Downloads/jw-cartoner-cover-1.snapshot.1$ dwg2dxf -v JW.dwg 
Reading DWG file JW.dwg
Warning: Unhandled Class object 501 MATERIAL (0x481) 14/0
Warning: Unhandled Class object 501 MATERIAL (0x481) 22/0
Warning: Unhandled Class object 501 MATERIAL (0x481) 30/0
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 39/0
ERROR: bit_read_BL: unexpected 2-bit code: '11'
ERROR: bit_read_BL: unexpected 2-bit code: '11'
ERROR: bit_read_TU buffer overflow at 38, length 256
ERROR: bit_read_TU buffer overflow at 38, length 256
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 40/0
ERROR: bit_read_BL: unexpected 2-bit code: '11'
ERROR: bit_read_BL: unexpected 2-bit code: '11'
ERROR: bit_read_TU buffer overflow at 39, length 256
ERROR: bit_read_TU buffer overflow at 39, length 256
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 41/0
ERROR: bit_read_BL: unexpected 2-bit code: '11'
ERROR: bit_read_BL: unexpected 2-bit code: '11'
ERROR: bit_read_TU buffer overflow at 39, length 256
ERROR: bit_read_TU buffer overflow at 39, length 256
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 42/0
ERROR: bit_read_BL: unexpected 2-bit code: '11'
ERROR: bit_read_BL: unexpected 2-bit code: '11'
ERROR: bit_read_TU buffer overflow at 39, length 256
ERROR: bit_read_TU buffer overflow at 39, length 256
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 43/0
ERROR: bit_read_TU buffer overflow at 20, length 153
ERROR: bit_read_BL: unexpected 2-bit code: '11'
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 44/0
ERROR: bit_read_BL: unexpected 2-bit code: '11'
ERROR: bit_read_TU buffer overflow at 23, length 250
ERROR: bit_read_BL: unexpected 2-bit code: '11'
ERROR: bit_read_TU buffer overflow at 35, length 256
ERROR: bit_read_TU buffer overflow at 36, length 256
ERROR: bit_read_BL: unexpected 2-bit code: '11'
ERROR: bit_read_TU buffer overflow at 39, length 2112
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 45/0
ERROR: bit_read_BL: unexpected 2-bit code: '11'
ERROR: bit_read_TU buffer overflow at 23, length 250
ERROR: bit_read_BL: unexpected 2-bit code: '11'
ERROR: bit_read_TU buffer overflow at 35, length 256
ERROR: bit_read_TU buffer overflow at 36, length 256
ERROR: bit_read_BL: unexpected 2-bit code: '11'
ERROR: bit_read_TU buffer overflow at 39, length 2112
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 46/0
ERROR: bit_read_TU buffer overflow at 23, length 256
ERROR: bit_read_TU buffer overflow at 25, length 59495
ERROR: bit_read_BL: unexpected 2-bit code: '11'
ERROR: bit_read_TU buffer overflow at 38, length 256
ERROR: bit_read_TU buffer overflow at 38, length 256
ERROR: bit_read_BL: unexpected 2-bit code: '11'
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 47/0
ERROR: bit_read_BL: unexpected 2-bit code: '11'
ERROR: bit_read_TU buffer overflow at 24, length 250
ERROR: bit_read_BL: unexpected 2-bit code: '11'
ERROR: bit_read_TU buffer overflow at 36, length 256
ERROR: bit_read_TU buffer overflow at 37, length 256
ERROR: bit_read_BL: unexpected 2-bit code: '11'
ERROR: bit_read_TU buffer overflow at 40, length 2112
Warning: Unstable Class object 502 VISUALSTYLE (0xfff) 48/0
ERROR: bit_read_TU buffer overflow at 23, length 256
ERROR: bit_read_TU buffer overflow at 25, length 59495
ERROR: bit_read_BL: unexpected 2-bit code: '11'
ERROR: bit_read_TU buffer overflow at 38, length 256
ERROR: bit_read_TU buffer overflow at 38, length 256
ERROR: bit_read_BL: unexpected 2-bit code: '11'
[... 222 lines cut ...]
Warning: Unknown Class entity 609 SURFACE (0xfff)
Warning: Unknown Class entity 609 SURFACE (0xfff)
Warning: Unknown Class entity 609 SURFACE (0xfff)
Warning: Unknown Class entity 609 SURFACE (0xfff)
Warning: Unknown Class entity 609 SURFACE (0xfff)
Warning: Unknown Class entity 609 SURFACE (0xfff)
Warning: Unknown Class entity 609 SURFACE (0xfff)
Warning: Unknown Class entity 609 SURFACE (0xfff)
Warning: Unknown Class entity 609 SURFACE (0xfff)
Warning: Unknown Class entity 609 SURFACE (0xfff)
Warning: Unknown Class entity 609 SURFACE (0xfff)
Warning: Unknown Class entity 609 SURFACE (0xfff)
Warning: Unknown Class entity 609 SURFACE (0xfff)
Warning: Unknown Class entity 609 SURFACE (0xfff)
Warning: Unhandled Class object 610 RAPIDRTRENDERSETTINGS (0x400) 1119/0
Warning: Unhandled Class object 610 RAPIDRTRENDERSETTINGS (0x400) 1120/0
Warning: Unhandled Class object 610 RAPIDRTRENDERSETTINGS (0x400) 1121/0
Warning: Unhandled Class object 610 RAPIDRTRENDERSETTINGS (0x400) 1122/0
Warning: Unhandled Class object 610 RAPIDRTRENDERSETTINGS (0x400) 1123/0
Warning: Unhandled Class object 610 RAPIDRTRENDERSETTINGS (0x400) 1124/0
Warning: Unhandled Class object 610 RAPIDRTRENDERSETTINGS (0x400) 1125/0
Warning: thumbnail.size mismatch: 5261 != 0
Writing DXF file JW.dxf

```

rurban, 2020-02-06:
Can the JW.dxf be imported back?

The warnings and errors are expected for those unknown classes. They are either ignore or partially tried to be decoded, and then skipped if unsuccessful.

rurban, 2020-02-06:
Vimal: Could you also provide a r2013 DXF for the JW.dwg? So we can compare against. I only did a teigha conversion, but this misses out on all the unknown classes, which you have loaded.

vimalpillai21, 2020-02-07:
I have donwloaded the file from https://grabcad.com/library/jw-cartoner-cover-1. I wanted to first convert this dwg file to dxf format and then to json format, So that I can view it in my web application using three.js .
[jw-cartoner-cover-1.snapshot.1.zip](https://github.com/LibreDWG/libredwg/files/4168897/jw-cartoner-cover-1.snapshot.1.zip)

rurban, 2020-02-07:
Thanks, I got now the normal objects as DXF from this file. Not much differences from teigha. 
I don't think that the extra applications create more, so it's fine. Working on it

rurban, 2020-02-07:
To the actual error: "TABLESTYLE r2010+ not yet implemented"
should say all. This object is ignored on all versions 2010 and higher.

closed 2020-02-07 by rurban