LibreDWG/libredwg #203: Make fails on Perl checks, which perl module is needed?

ontoast, 2020-02-17:
This post is listed in the order of questions first followed by specs, problem descriptions, and information.

### Questions:
1. What perl module is needed to get past the make check error shown below? 
    (see the "04 output:" of the "Description of the problem:" section shown below.)
2. What does the error message about the undefined symbol in line 187 of the DynaLoader.pm file mean and is this significant or can this be ignored for now?
    (see the "Make check error examined:" section near the bottom)
3. Please let me know if I can provide anything else that might be helpful for figuring this out. Thanks.

### Specs:
**OS:** Linux
**Desktop:** Xubuntu 18.04.4 LTS
**Bash version:** 4.4.20(1)-release (x86_64-pc-linux-gnu)
**Perl version:** 5.26.1-6ubuntu0.3
**Dependencies installed:**
- TeXinfo for building the docs
- pslib for dwg2ps: http://pslib.sourceforge.net/doc/pslib.html
- pcre2 with 8bit and 16bit libraries for dwggrep regexp support
- SWIG 1.7+
- Python 2.7 or 3.x development headers (debian: python-dev; yum: python-devel)
  and the libxml2 python bindings.
- Perl 5 for the perl bindings
- doxygen to generate the reference manual
- jq to check json validity
- valgrind to find leaks and memory bugs
- timeout to help tests with large or hanging DWG's
- GNU parallel to speed up special tests

**Dependencies not installed:**
- picat

The above package was not installed due to not being available by default in the apt-cache repositories.

### Description of the problem:
**Steps taken:**
**01 command:** sh ./autogen.sh
**01 outcome:** no errors.

**02 command:** ./configure
**02 outcome:** no errors, config looked ok.

**03 command:** make
**03 outcome:** Two possible errors listed, both with the same message:

> make regen-dynapi might be needed

***Here is the truncated make output:***
- Paths and successful make blocks have been truncated for readability.
- Make message blocks are shown in the order received.
- Note that "all-am" passes the "make regen-dynapi might be needed" message, but then completes successfully without this error in a later message block.
- Note that "perl" shows no errors.

```Make
make  all-recursive
...
make  all-am
make[3]: Entering directory '.../libredwg/src'
echo make regen-dynapi might be needed
make regen-dynapi might be needed
...
Making all in doc
make[2]: Entering directory '.../libredwg/doc'
echo make regen-dynapi might be needed
make regen-dynapi might be needed
...
make  all-am
make[4]: Entering directory '.../libredwg/bindings/python'
make[4]: Nothing to be done for 'all-am'.
make[4]: Leaving directory '.../libredwg/bindings/python'
make[3]: Leaving directory '.../libredwg/bindings/python'
[... 142 lines cut ...]
    # in this perl code simply because this was the last perl code
    # it executed.

    my $flags = $module->dl_load_flags;
    
    my $libref = dl_load_file($file, $flags) or
	croak("Can't load '$file' for module $module: ".dl_error());

    push(@dl_librefs,$libref);  # record loaded object

    $boot_symbol_ref = dl_find_symbol($libref, $bootname) or
         croak("Can't find '$bootname' symbol in $file\n");

    push(@dl_modules, $module); # record loaded module

  boot:
    my $xs = dl_install_xsub("${module}::bootstrap", $boot_symbol_ref, $file);

    # See comment block above

	push(@dl_shared_objects, $file); # record files loaded

    &$xs(@args);
}
```

ontoast, 2020-02-17:
### UPDATE:

A package was discovered in the default apt-cache named:
libconvert-binary-c-perl

The above named package was installed, however, this made no difference after "make" and "make check". The same error as before still persists.

rurban, 2020-02-17:
1. What perl module is needed to get past the make check error shown below?
(see the "04 output:" of the "Description of the problem:" section shown below.)

None. The problem loading a non-matching older shared library.

Delete any older libredwg shared library from your system, which is loaded before the current one.

ontoast, 2020-02-17:
Thanks rurban.

I get the following from ldd:
```
ldd ./bindings/perl/blib/arch/auto/LibreDWG/LibreDWG.so
	linux-vdso.so.1 (0x00007ffdf8bfa000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f648ba8f000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f648ca44000)
````
I'm researching how to track this down now. I'll see if I can find someone who's done this before.

Thanks again for pointing me in the right direction.

rurban, 2020-02-17:
ldd doesn't help, the so is loaded at runtime via dl_load. export PERL_DL_DEBUG=5 should help.

ontoast, 2020-02-17:
Thanks again. Very helpful. 

Out of curiosity I checked the latest pre-release package [0.10.1.2905](https://github.com/LibreDWG/libredwg/releases/tag/0.10.1.2905) but it failed on configure. However this is off topic and would need its own issue tracking.

I'm going back to testing the github clone using the environment variable you've suggested now.

ontoast, 2020-02-17:
### Update:
After, export PERL_DL_DEBUG=5, and after "make check" on a clone of this mornings github for libredwg, there is a warning about a missing "swig_perl.c" (see below), and a more descripting fail message.

### Question:
- How do I identify if a library is "old" in the context of what LibreDWG requires?

### Latest "make check" result:
```
...
make[2]: Entering directory '~/Programs/libredwg/doc'
echo make regen-dynapi might be needed
make regen-dynapi might be needed
...
make[3]: Entering directory '~/Programs/libredwg/bindings'
cd perl && \
  /usr/bin/perl Makefile.PL PERL="/usr/bin/perl" INSTALLDIRS=site
Checking if your kit is complete...
Warning: the following files are missing in your kit:
	swig_perl.c
Please inform the author.
...
make[4]: Entering directory '~/Programs/libredwg/bindings/perl'
/usr/bin/perl -MExtUtils::Command::MM -e 'cp_nonempty' -- LibreDWG.bs blib/arch/auto/LibreDWG/LibreDWG.bs 644
PERL_DL_NONLAZY=1 "/usr/bin/perl" "-MExtUtils::Command::MM" "-MTest::Harness" "-e" "undef *Test::Harness::Switches; test_harness(0, 'blib/lib', 'blib/arch')" t/*.t
DynaLoader.pm loaded (/etc/perl /usr/local/lib/x86_64-linux-gnu/perl/5.26.1 /usr/local/share/perl/5.26.1 /usr/lib/x86_64-linux-gnu/perl5/5.26 /usr/share/perl5 /usr/lib/x86_64-linux-gnu/perl/5.26 /usr/share/perl/5.26 /usr/local/lib/site_perl /usr/lib/x86_64-linux-gnu/perl-base, /usr/local/lib /usr/lib/gcc/x86_64-linux-gnu/7/include-fixed /usr/include/x86_64-linux-gnu /usr/lib /lib/x86_64-linux-gnu /lib/../lib /usr/lib/x86_64-linux-gnu /usr/lib/../lib /lib ~/Programs/libredwg/src/.libs)
DynaLoader::bootstrap for Time::HiRes (auto/Time/HiRes/HiRes.so)
t/0_basic.t .. DynaLoader.pm loaded (~/Programs/libredwg/bindings/perl/blib/lib ~/Programs/libredwg/bindings/perl/blib/arch /etc/perl /usr/local/lib/x86_64-linux-gnu/perl/5.26.1 /usr/local/share/perl/5.26.1 /usr/lib/x86_64-linux-gnu/perl5/5.26 /usr/share/perl5 /usr/lib/x86_64-linux-gnu/perl/5.26 /usr/share/perl/5.26 /usr/local/lib/site_perl /usr/lib/x86_64-linux-gnu/perl-base ., /usr/local/lib /usr/lib/gcc/x86_64-linux-gnu/7/include-fixed /usr/include/x86_64-linux-gnu /usr/lib /lib/x86_64-linux-gnu /lib/../lib /usr/lib/x86_64-linux-gnu /usr/lib/../lib /lib ~/Programs/libredwg/src/.libs)
DynaLoader::bootstrap for mro (auto/mro/mro.so)
DynaLoader::bootstrap for LibreDWG (auto/LibreDWG/LibreDWG.so)
t/0_basic.t .. 1/1 
#   Failed test 'use LibreDWG;'
#   at t/0_basic.t line 2.
#     Tried to use 'LibreDWG'.
#     Error:  Can't load '~/Programs/libredwg/bindings/perl/blib/arch/auto/LibreDWG/LibreDWG.so' for module LibreDWG: ~/Programs/libredwg/bindings/perl/blib/arch/auto/LibreDWG/LibreDWG.so: undefined symbol: dwg_get_entity_layer at /usr/lib/x86_64-linux-gnu/perl/5.26/DynaLoader.pm line 187.
#  at ~/Programs/libredwg/bindings/perl/blib/lib/LibreDWG.pm line 11.
# Compilation failed in require at t/0_basic.t line 2.
# BEGIN failed--compilation aborted at t/0_basic.t line 2.
# Looks like you failed 1 test of 1.
t/0_basic.t .. Dubious, test returned 1 (wstat 256, 0x100)
Failed 1/1 subtests 
...
```

***P.S.*** Thanks again for all the help. I know your busy, and I really appreciate your time. Thank you.

rurban, 2020-02-17:
old is a version without dwg_get_entity_layer, as the error message says. So the correct binding is dlloaded, auto/LibreDWG/LibreDWG.so, and this loads the main so, in your case a version without dwg_get_entity_layer. There should be some GLIBC MALLOC debugging var to see which dependent so is loaded.

ontoast, 2020-02-17:
Thank you for the clarification and again for pointing me to something I can read more about.

I must return to work and will most likely not be able to respond again until Friday. Thank you again for your help and support.

ontoast, 2020-03-07:
Apologies for any one reading this thread, but I do not have time to track this down due to work hours. Thank you to Rurban for pointing me in a helpful direction. Keep up the great work! :)

closed 2020-03-07 by ontoast