LibreDWG/libredwg #209: GeoJSON exporting : error in json format

yossefazoulay, 2020-03-10:
When using the dwgread command and export a dwg to json format, the json exported add an extra comma which is a non-valid json :

{
	"type": "FeatureCollection",
	"features": [{
			"type": "Feature",
			"properties": {
				"SubClasses": "AcDbObject : AcDbGeoData",
				"EntityHandle": "4"
			} , <===
		}, 
}

yossefazoulay, 2020-03-10:
another issue is that no error when exporting to GeoJSON even if dwg is corrupted : 

{
  "type": "FeatureCollection",
  "features":   ], <====
  "geocoding":   {
    "creation_date": "2020-03-09",
    "generator":     {
      "author":       {
        "name": "dwgread"
      },
      "package": "LibreDWG",
      "version": "0.10.1.364_953cc"
    }
  }
}

rurban, 2020-03-10:
Can you send me the DWG for these errors? Needs to be fixed.

yossefaz, 2020-03-11:
Here is a Google drive link to two dwg files :

2010_1.dwg => for the first error (extra comma)

2010_2.dwg => for the second error (corrupted file, will export to json anyway)

I also joined here the output json file

https://drive.google.com/drive/folders/1NRqt1OARxFYXFybML1uv7Ze1f6ElHc-O?usp=sharing

rurban, 2020-03-11:
Wonderful! The first testcase with an GEODATA object. Thanks a lot.

closed 2020-03-14 by rurban