LibreDWG/libredwg #218: installing libredwg on windows

carlopav, 2020-03-29:
Hello @rurban,
sorry for the simple question: i'm a complete newby. I downloaded the windows nightly at https://github.com/LibreDWG/libredwg/releases.
But I have no idea on where i should put the files to make them available in FreeCAD.
Any avice?
thx :)

carlo

rurban, 2020-03-29:
Nope, haven't tried the new FreeCAD yet. Please ask over there.

carlopav, 2020-04-01:
I succeded opening it from FreeCAD, but it gives me the same error that occurs when I try to execute it from windows command prompt:
"Cannot start program because libssp-0.dll is not present on the computer. Try to re-install the program".
I succeded to run it only from "Anaconda Prompt (Anaconda3)".
Any advice?

rurban, 2020-04-01:
Try to find the matching mingw libssp DLL and copy it to the DWG exe's. I might include it later into the zip. Mingw frequently changes deps.

carlopav, 2020-04-01:
Great! I downloaded the DLL and placed into the exe's folder and it works.

closed 2020-04-06 by rurban