LibreDWG/libredwg #229: dwgplot via GNU Plotutils

rurban, 2020-04-15:
Then we can deprecate dwg2SVG and dwg2ps, and rather use plotutils, either via libplot or via plot(1),which supports much more output formats, bitmap and vector. Even real plotting formats, like pcl5 and hpgl/2. But also png, svg, ps, ...

The current idea is to be independent on libplot, and generate ASCII plot(5) format manually,
or if available via the libplot C interface.

matndev, 2021-04-09:
Any news ? Very interested in converting dwg to jpg

rurban, 2021-04-10:
No, this would need a lot of work. See the `work/dwgplot` branch

paolo-caroni, 2021-05-16:
This is a very good idea, have you planned to support PlotStyle (.ctb and .stb)?
I'm writing specification of these type of files that maybe can be interesting, also a simple C PoC for read and edit it.
https://github.com/paolo-caroni/LibrePIA
For the spec see doc folder.