LibreDWG/libredwg #231: Does dwgread already support ps (postscript?)

Sicos1977, 2020-04-29:
I was looking on the internet for an app that could convert DGW to PDF and found dwgread. I allready am using Ghostscript so it would be easy for me to convert the ps file to PDF.

But whenever I try dwgread complains that test.ps is an invalid file.

Sicos1977, 2020-04-29:
An SVG file would also something that I can use but when I convert my DWG file to SVG it produces a blank page when I view it in Chrome.

I can sent you the DWG file through mail if you want but I can't upload it here because it is from a client.

rurban, 2020-04-29:
dwgread will not support PS, you need to use dwg2ps. But it handles only a few entities, SVG more.
I will convert those to one dwgplot, which can then produce many more output formats, bitmap and vector.

closed 2020-05-05 by rurban