LibreDWG/libredwg #233: ./configure error.

gergul, 2020-05-27:
config.log:
`This file contains any messages produced by compilers while
running configure, to aid debugging if configure makes a mistake.

It was created by LibreDWG configure Usage: build-aux/git-version-gen OPTION... ./.tarball-version TAG-NORMALIZATION-SED-SCRIPT
Print a version string.

Options:

   --prefix PREFIX    prefix of git tags (default 'v')
   --fallback VERSION
                      fallback version to use if "git --version" fails

   --help             display this help and exit
   --version          output version information and exit

Running without arguments will suffice in most cases.
, which was
generated by GNU Autoconf 2.69.  Invocation command line was

  $ ./configure 

## --------- ##
## Platform. ##
## --------- ##

hostname = gergul-centos-vm
uname -m = x86_64
uname -r = 3.10.0-1127.8.2.el7.x86_64
uname -s = Linux
uname -v = #1 SMP Tue May 12 16:57:42 UTC 2020

/usr/bin/uname -p = x86_64
/bin/uname -X     = unknown

/bin/arch              = x86_64
/usr/bin/arch -k       = unknown
/usr/convex/getsysinfo = unknown
/usr/bin/hostinfo      = unknown
/bin/machine           = unknown
/usr/bin/oslevel       = unknown
/bin/universe          = unknown

PATH: /usr/local/bin
PATH: /usr/local/sbin
PATH: /usr/bin
PATH: /usr/sbin
PATH: /bin
PATH: /sbin
PATH: /home/gergul/.local/bin
PATH: /home/gergul/bin

## ----------- ##
## Core tests. ##
## ----------- ##

configure:2875: checking for a BSD-compatible install
configure:2943: result: /usr/bin/install -c
configure:2954: checking whether build environment is sane
configure:3009: result: yes
configure:3160: checking for a thread-safe mkdir -p
configure:3199: result: /usr/bin/mkdir -p
configure:3206: checking for gawk
configure:3222: found /usr/bin/gawk
configure:3233: result: gawk
configure:3244: checking whether make sets $(MAKE)
configure:3266: result: yes
configure:3295: checking whether make supports nested variables
configure:3312: result: yes
configure:3462: checking whether make supports nested variables
[... 454 lines cut ...]
   --prefix PREFIX    prefix of git tags (default v)
   --fallback VERSION
                      fallback version to use if "git --version" fails

   --help             display this help and exit
   --version          output version information and exit

Running without arguments will suffice in most cases.
"
#define PACKAGE_BUGREPORT "libredwg@gnu.org"
#define PACKAGE_URL "https://savannah.gnu.org/projects/libredwg/"

configure: exit 77`

My automake version:
`automake (GNU automake) 1.14.1
Copyright (C) 2013 Free Software Foundation, Inc.
License GPLv2+: GNU GPL version 2 or later <http://gnu.org/licenses/gpl-2.0.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Written by Tom Tromey <tromey@redhat.com>
       and Alexandre Duret-Lutz <adl@gnu.org>.`

Thanks.

rurban, 2020-05-27:
Can you run `sh autogen.sh`

The version script has broken data somehow

gergul, 2020-05-28:
Thank you for your answer.  I got rid of it with git clone again.

closed 2020-05-28 by gergul