LibreDWG/libredwg #234: How to get the owner block of each entities.

gergul, 2020-06-05:
I parse a dwg file which has some entities in *Paper_Space. I want to get entities owner block
`void printOwner(Dwg_Object* obj, Dwg_Data* dwg)
{
	Dwg_Object *owner = NULL;
	if (is_dwg_entity(obj->name) && obj->tio.entity->ownerhandle)
	{
		owner = dwg_resolve_handle(dwg, obj->tio.entity->ownerhandle->absolute_ref);
	}
	else if (is_dwg_object(obj->name) && obj->tio.object->ownerhandle)
	{
		owner = dwg_resolve_handle(dwg, obj->tio.object->ownerhandle->absolute_ref);
	}
	else
	{
		printf("owner_block=*Model_Space\n");
	}

	if (owner)
	{
		if (owner->type == DWG_TYPE_BLOCK_HEADER)
		{
			Dwg_Object_BLOCK_HEADER* bhr = owner->tio.object->tio.BLOCK_HEADER;
			int needfree = 0;
			char* szName = convertString(dwg->header.version, bhr->name, &needfree);// convert str use bit_convert_TU
			printf("owner_block=%s\n", szName);
			if (needfree)
				free(szName);
		}
	}
}`
But I found the entities in *Paper_Space will print "owner_block=*Model_Space".
How shoud I get the owner block of each entities.
Thank you.

rurban, 2020-06-16:
Sorry, never answered

gergul, 2020-06-22:
Thank u.

closed 2020-06-22 by gergul